VCL3D/VolumetricCapture #1: Is Copper Wire OK?

vhanded, 2019-05-24:
Firstly, thanks for this very detail tutorial, the most complete yet.

I accidentally bought copper wire for the HW sync cable, but in tutorial you mentioned twisted pair. Is it ok to use copper wire?
[image]

Thanks for your advice.

ankarako, 2019-05-24:
Hi @vhanded ,

We haven't tested HW syncing with not shielded solid copper wire cables. We followed Intel's white papers ([paper1](https://realsense.intel.com/wp-content/uploads/sites/63/Intel_RealSense_Multi-camera_Webinar.pdf), [paper2](https://realsense.intel.com/wp-content/uploads/sites/63/Multiple_Camera_WhitePaper04.pdf)) that recommend shielded cables for long cable lengths.

From my experience, I think that it would probably be OK if you don't have long cables. You can also check the integrity of the synchronization signal if you have access to an oscilloscope.

Best,
Antonis

vhanded, 2019-06-03:
Another question, the white paper mentioned that to use resistor of 20k, but you 2 x 10k instead. Is it because of resistor / transistor size limitation?

ankarako, 2019-06-04:
Hi @vhanded

We use 2x10K resistors connected in series (which makes a total of 20K), and that's because the electronic component stores (in general) don't have available 20K resistors out of the box. (Typical resistor values are 10K, 22K etc. Not that you won't find 20K resistors anywhere, but it's not that common). So the low-pass filter is exactly the same as Intel's papers.

Please, don't hesitate to ask anything further.
Best,
Antonis

vhanded, 2019-06-11:
Hi @ankarako 

I do not have 4 Intel NUC in my setup, all 4 sensors are attached to a USB3 hub. And, I am working in Ubuntu, so might need to compile the source code in linux.

But, I am not able to find the source code. Is it publish somewhere?

Thanks.

ankarako, 2019-06-12:
Hi @vhanded ,

Unfortunately, we do not provide the source code of the system. We provide a distributed solution for managing a multi-view rgb-d capturing system.

closed 2019-06-18 by ankarako