VCL3D/VolumetricCapture #2: Can volumetric_capture.exe detect realsense via USB?

vhanded, 2019-06-17:
I dont have Intel NUC or any extra PC and switch for connection, so I can just connect all my devices using USB3 hub.

Is this setup supported?

Thanks.

vladsterz, 2019-06-18:
Hello @vhanded , 
I assume you have installed RabbitMQ on the PC you are going to run VolumetricCapture. Next step is to run the [remote_eye](https://github.com/VCL3D/VolumetricCapture/wiki/Remote-Service) executable, which handles the device functionality, as well as capturing, serializing and sending the captured data across network. The wiki section refers to the __remote_eye service__ , but can manually run the executable under __bin__ folder in the downloaded __remote_eye__ folder, given the address of the RabbitMQ broker as _remote_eye.exe --address [address of the broker] --user [username of the broker] --pass [password of the broker]_.
Best, Vlad

vhanded, 2019-06-27:
Since I dont have slave PC / Intel NUC, I plugged 1 of my Realsense to my main PC.

I start remote eye like this:
`remote_eye.exe --address 2182`

But I got the following error:
[image]

However, I can start the service successfully:
```
C:\Capturer\remote_eye_service>net start remote_eye_service
The remote_eye_service service is starting.
The remote_eye_service service was started successfully.
```

While performing device registration: 
`device_repository.exe add 813612020030`

I got:
`2019-06-27 09:55:56.840 FATAL [3440] [dvc::IO::read@111] File doesn't exist.`

Any idea?

vladsterz, 2019-06-27:
Hello once again @vhanded.
Firstly, argument --address refers to the IP address of the RMQ.
Please check if service does indeed starts a remote_eye.exe instance, or it does fails quietly.
At last, this **FATAL** shouldn't be a fatal message and you should ignore it, the file __device_repository.json__ should be create in the calling directory.

vhanded, 2019-06-27:
Thanks for reply, @vladsterz 
So I started remote_eye: `net start remote_eye_service`

For device_repository.exe, It still show File doesn't exist. So in the same directory with device_repository.exe, I created a file name `device_repository.json`, with empty array. Then the error is gone. However, new device is not added into the json file.

My command:
```
C:\Users\xxxxx\Downloads\VolumetricReleasex64>device_repository.exe add M1

C:\Users\xxxxx\Downloads\VolumetricReleasex64>
```

No proper output from this command.

vladsterz, 2019-06-28:
Hello once again,
So I downloaded the latest release and just run the software in the same way as you did and it works.
[image]
Recently me updated the Release v.2 , maybe you should re-download it.

vladsterz, 2019-09-26:
Hello @vhanded , do you still need any support for this?

vladsterz, 2019-11-08:
Closing this for now.

closed 2019-11-08 by vladsterz