VCL3D/VolumetricCapture #3: Device Repository - executable not found

manouj, 2019-06-20:
Hi,
I am unable to find the device_repository.exe file in the extracted VolumetricReleasex64.rar file folder. I can only see the device_repository.json file.

vladsterz, 2019-06-21:
Hello @manouj , 
you are utterly correct. We are going to upload it within the newest Release, which is going to be soon.
This issue will be closed, once the new Release is been uploaded.
Thank you and stay tuned.

manouj, 2019-06-21:
Hi,

Thank you so much for your prompt response. This is exciting and we see great potential in this.
I am eager to know when the new release will be uploaded. Also, kindly let me know whether it is possible to extend this setup to 8 cameras.

Thanks again,
Manouj

vladsterz, 2019-06-21:
This is kind of hard to answer since our team is busy with some deadlines.
A wild guess would be until the start of the next week.

manouj, 2019-06-21:
Great. Thank you.

@grinfax

spthermo, 2019-06-24:
Hi @manouj 

you can use the updated Release v2.0 that contains the device_repository.exe. This closes #3

closed 2019-06-24 by spthermo

manouj, 2019-06-24:
Thank you so much! I really appreciate your efforts. 
Please let me know if I can connect 8 cameras using this setup. 
Also, what will be the output file format of the recordings. Can I import the recordings into Unity?

Once again, thank you very much for providing this.

vladsterz, 2019-06-24:
Hello once again @manouj ,
We haven't looked into any setup consisting of more than 4 devices, but we intend to. You can always try **BUT**, calibration functionality won't work since it is designed to work with 4 devices, based on [this](https://vcl.iti.gr/publication/markerless-structure-based-multi-sensor-calibration-for-free-viewpoint-video-capture/) work (and further extended in the published paper of the [present system](https://vcl.iti.gr/publication/a-low-cost-flexible-and-portable-volumetric-capturing-system/) ).
This feature is going to be updated, it is one of our research directions, but I cannot guarantee on the date of the release.
Saving format is (briefly) described [here ](https://vcl.iti.gr/publication/a-low-cost-flexible-and-portable-volumetric-capturing-system/).
Importing data into Unity is not provided as an option so far and is not on our TODO lists so far.
Hope I helped, don't hesitate to ask for any further information.

Vlad

manouj, 2019-06-24:
Thanks @vladsterz 
I missed this in the first place. File format details are available in the below link
https://github.com/VCL3D/VolumetricCapture/wiki/File-Format

I understood that 2.5D player is used for playback of .cdv files. But I could not find the installation files of the player. Please let me know where I can find the installation files of 2.5D player.