VCL3D/VolumetricCapture #4: synch circuit

lsorgi, 2019-07-05:
hi there, 

I want to realize a synchronized multicamera system as described in this repo.
the question might seem naive...but I am trying to understand the synchronization circuit as it is described in wiki

https://github.com/VCL3D/VolumetricCapture/wiki/Synchronization-Cables

and but I can't figure out how this can be equivalent to what is shown on the Intel white paper  

https://realsense.intel.com/wp-content/uploads/sites/63/Intel_RealSense_Multi-camera_Webinar.pdf 

would you please clarify why in the wiki instruction every camera has 2 resistors and 1 capacitor and why the connection are different?

thanks a lot

ankarako, 2019-07-06:
Hi @lsorgi ,

Actually, the circuits are just the same. Generally, a circuit schematic and its actual implementation on a PCB  may differ in terms of "looks". It is not always easy to achieve the same circuit connections as they are depicted by a circuit schematic. 
Apart from this, Intel's circuit's about a very simple RC lowpass filter with a 20K resistor and a 22nF capacitor connected in parallel. Ours is just the same. The only difference is that we use two 10K resistors (connected in series which add up to 20K).

Don't hesitate to ask anything further. (Please check the discussion [here](https://github.com/VCL3D/VolumetricCapture/issues/1) too)
Regards,
Antonis

lsorgi, 2019-07-06:
Hi!
... ah sure! I didn’t pay attention and missed the fact that they are 2 10k resistors 😁
Thanks for the fast reply

closed 2019-07-10 by zuru