VCL3D/VolumetricCapture #8: Device_repository needs more info on wiki

MarcKami, 2019-10-08:
The wiki of device_repository needs to include a step by step explanation of how to create the final json file and where to put it (all the folders).
I notice when you put only on root folder the calibration step doesn't find the intrinsecs, but when you put it on ~/Resources and ~/Data/Executables/Calibration it works perfectly. Maybe you only need to put in root and one of those folders, but I noticed you already have the same json inside in both of them.

MarcKami, 2019-10-08:
btw, the device_repository exe and dll would be better to put it in a folder to make it clear. But is not an issue after all.

vladsterz, 2019-10-25:
Dear @MarcKami , 
A new release (3.0) has been updated, with the corresponding wiki page update.

closed 2019-10-25 by vladsterz