VCL3D/VolumetricCapture #10: Remote eye service is not working

jbyu, 2019-11-07:
I follow the instruction: installed RabbitMQ and open 5672,11234 port on work station and min-PCs. 
Download remote_eye.zip and extract to C:/Capturer, then install service with administrator.
But the devices won't show up.
I use Volumetric Capture and Tools Release 3.0 and remote_eye.zip 2.0
Thank you.

ankarako, 2019-11-07:
Hi @jbyu , did you install the Service on all the mini-PCs?

jbyu, 2019-11-07:
I just tested one mini-PC.
1. The service is running.
2. The log shows no request from the workstation. It only shows
[timestamp]: Listener started listening...
[timestamp]: Process name: C:/Capturer/bin/remote_eye.exe
3. I can manually open remote_eye and connect to the workstation.

ankarako, 2019-11-07:
Did you open port 11234 on UDP or TCP protocol? (It should be UDP, just saw that in the wiki this is omitted)

jbyu, 2019-11-08:
Yes, UDP is the key! Thank you.

closed 2019-11-08 by jbyu

ankarako, 2019-11-08:
Happy to help.