VCL3D/VolumetricCapture #11: MultiView Player doesn't work

jbyu, 2019-11-12:
Hi, I recorded a cdv file, but can't play it with multi-view player.
When I drop the file, it shows "Color dt:0 Depth dt:8067672801865 Frame ID:-1".
I don't know how to check the cdv files.
Thank you.

vladsterz, 2019-11-12:
Hello @jbyu ,
always glad to see people using our solution. 
Does the .cdv file has any _meaningful size_ ?
Can I have a screenshot our your multi-view player?
Thx, Vlad

jbyu, 2019-11-12:
Yes, I recorded 8 sec data about 29MB.
[image]

vladsterz, 2019-11-12:
So when you press the play button in the control panel, it does not playback your file?

vladsterz, 2019-11-12:
If your recording does not contain sensitive data, could you please send me so I can reproduce the error?

jbyu, 2019-11-12:
After maximizing the window, I found the control panel and its playable.
Sorry for bothering you, my bad.
BTW, How to reconstruct captured data to mesh? 
Could you give some advise?
Thank you.

vladsterz, 2019-11-13:
In general the GUI is not pinned as you probably found out, and the positions of the windows are saved to the corresponding _imgui.ini_ file. 
Our lab's work on real time reconstruction can be found [here](https://vcl.iti.gr/publication/an-integrated-platform-for-live-3d-human-reconstruction-and-motion-capturing/) and you should definitely check [holoportation](https://www.researchgate.net/publication/306544236_Holoportation_Virtual_3D_Teleportation_in_Real-time).
If your recordings are [calibrated](https://github.com/VCL3D/VolumetricCapture/wiki/Volumetric-Capture-Application#system-calibration) with our [structure](https://github.com/VCL3D/VolumetricCapture/wiki/Calibration-&-Setup), you can dump you pointclouds from Multi-View Player with the _calibrated_pointcloud_ option checked.

jbyu, 2019-11-14:
1. There is no timeline panel, do I miss something?
[image]

2. it shows warnings about intrinsic value, but I did have device_repository.json. How to fix it?
[image]

3. Is there any software to do reconstruction?
Thank you.

vladsterz, 2019-11-14:
Hello once again.
1) Timeline window shows up once temporal synchronization is done. It is a qualitative metric in order to assess the result of the calibration. You can tune the settings under synchronization GUI and press _synchronize_ to start the procedure. In order for this to work, before every recording you have to compute the synchronization offsets for every device is it is shown [here](https://github.com/VCL3D/VolumetricCapture/blob/master/doc/synchronization.gif).
2) I assume you have created a _device_repository.json_ file using [this](https://github.com/VCL3D/VolumetricCapture/wiki/Device-Repository-Tool) tool. This error is a known to us issue which will be fixed in a later release. You can work around it by placing your _device_repository.json_ file under volumetric_capture\Data\Executables\Calibration . This must do the trick.
3) No, we do not provide a tool for 3D reconstruction.
Vlad

jbyu, 2019-11-15:
I found recording with (RGB - VGA, D - FHD / 4) can not dump calibrated data,
but (RGB - FHD, D - FHD / 4) does.

[image]

vladsterz, 2019-11-15:
Hello @jbyu ,
Thank you for pointing it out, this as well is going to be fixed in the upcoming Release. Do you require to have VGA recording?
Vlad

closed 2019-11-22 by jbyu