VCL3D/VolumetricCapture #17: Intel D415's stream fails to start

mhudnell, 2020-04-09:
In the volumetric_capture.exe, after hitting "Connect" to device, the stream fails to start, resulting in blank camera previews (see attached img of UI). The streams work occasionally, but this bug happens most of the time.

[image]

Upon inspecting the remote_eye.log on the mini-pcs, an error "No frames received, resetting device hardware ..." is given. Here is the full log of the incident:

> 2020-04-09 11:26:26.892 INFO  [2688] [main@76] Logging to: C:\Capturer\bin\remote_eye.log
2020-04-09 11:26:27.435 INFO  [2688] [main@87] Connected RS2 device /w serial no: 928222063567
2020-04-09 11:26:27.435 INFO  [2688] [main@92] Connecting to RabbitMQ broker: amqp://volumetric:capture@169.254.176.0:5672 # Handshake
2020-04-09 11:26:28.454 INFO  [2688] [remote::handshake::say_hello@49] Sent hello from 928222063567
2020-04-09 11:26:28.454 INFO  [2688] [main@230] Waiting for welcome (press 'q' to quit) ...
2020-04-09 11:26:32.109 DEBUG [9208] [main::<lambda_241a8e4d42eec98c02aa0213c5be8deb>::operator ()@119] Received Welcome:
{ msg = Welcome , serial = 928222063567 /w stream channel = d415_0_c1280_x_720_d320_x_180_stream /w params channel = d415_0_params /w workflow channel = d415_0_workflow /w info channel = d415_0_info /w preset = accuracy-100um /w sync type = Default
}
2020-04-09 11:26:32.110 INFO  [9208] [main::<lambda_241a8e4d42eec98c02aa0213c5be8deb>::operator ()@131] Creating networking components...
2020-04-09 11:26:32.111 INFO  [9208] [main::<lambda_241a8e4d42eec98c02aa0213c5be8deb>::operator ()@139] Streaming channel opened.
2020-04-09 11:26:32.117 ERROR [9208] [local::led_controller::led_controller@59] NUC Led controller app is not available @ C:\Capturer\bin\intel_nuc_led_utils.exe.
2020-04-09 11:26:32.117 INFO  [9208] [main::<lambda_241a8e4d42eec98c02aa0213c5be8deb>::operator ()@149] Command channel opened.
2020-04-09 11:26:32.118 INFO  [9208] [main::<lambda_241a8e4d42eec98c02aa0213c5be8deb>::operator ()@161] Initializing syncer on C:\Capturer\bin\PTPSync_slave.exe ...
2020-04-09 11:26:32.118 INFO  [9208] [main::<lambda_241a8e4d42eec98c02aa0213c5be8deb>::operator ()@163] Syncer initialized.
2020-04-09 11:26:32.119 INFO  [9208] [main::<lambda_241a8e4d42eec98c02aa0213c5be8deb>::operator ()@167] Initializing device...
2020-04-09 11:26:32.119 INFO  [9208] [main::<lambda_241a8e4d42eec98c02aa0213c5be8deb>::operator ()@174] Loading preset <accuracy-100um> from C:\Capturer\bin\Resources\accuracy-100um.json
2020-04-09 11:26:32.426 INFO  [9208] [main::<lambda_241a8e4d42eec98c02aa0213c5be8deb>::operator ()@178] Acquiring device parameters...
2020-04-09 11:26:34.268 INFO  [9208] [main::<lambda_241a8e4d42eec98c02aa0213c5be8deb>::operator ()@187] Device parameters initialized from configuration file: C:\Capturer\bin\Resources\accuracy-100um.json
2020-04-09 11:26:34.555 INFO  [9208] [main::<lambda_241a8e4d42eec98c02aa0213c5be8deb>::operator ()@190] Device started.
2020-04-09 11:26:34.555 INFO  [9208] [main::<lambda_241a8e4d42eec98c02aa0213c5be8deb>::operator ()@215] Device parameters session created.
2020-04-09 11:26:34.556 INFO  [9208] [main::<lambda_241a8e4d42eec98c02aa0213c5be8deb>::operator ()@221] Encoding /w compression params:
	JPEG:	 { Quality = 70 , Subsampling = 1 , Format = 1 , Fast DCT = on } 
	ENTROPY:	 { Type = 3 , Level = 4 , Shuffle = 1 } 
2020-04-09 11:26:34.556 INFO  [9208] [main::<lambda_241a8e4d42eec98c02aa0213c5be8deb>::operator ()@226] Streaming pipeline ready.
2020-04-09 11:26:35.062 ERROR [3452] [aqzn::realsense2::rs2rgbd_pipeline_autorecov_producer::reset@288] No frames received, resetting device hardware ...

ankarako, 2020-04-09:
Hi Max,

Could you check your devices' firmware, and make sure they are updated to the latest version?

Antonis

zuru, 2020-04-09:
Hi Max, 

usually when this happens, please switch the D415 device to another USB hub.

If you are using Intel NUCs, move it from the front hubs to the back, this ensures a controller switch and a restart of the device resources.

Another potential issue might be that the pin connectors for hardware sync are not properly done / mounted.

Let us know how it goes.

mhudnell, 2020-04-09:
Thanks for the responses, I updated the firmware, but the issue still occurs. After messing around and switching usb hubs / try refreshing / connecting a few times, I am able to get it to work eventually, so that works for now.

I haven't connected a hardware sync cable yet, so I don't know if that could be an issue.

vladsterz, 2020-04-09:
@mhudnell , 
sadly we've experienced the same issue with our RealSense DXXX series. Usually one (ore more) device(s) fail to start streaming, so what we do is change its USB port, select the corresponding device in "Connected Devices" menu in Volumetric Capture, hit refresh and the connect. It is not that much trouble usually, most of the times we do this procedure once. 
Enjoy our application :)

closed 2020-04-18 by zuru