VCL3D/VolumetricCapture #18: volumetric_capture.exe 'calibrate' fails to locate directories

mhudnell, 2020-04-09:
I'm following the directions from the wiki on how to calibrate. Upon clicking **Calibration > Calibrate**, I see the following errors saying the directories / files could not be found:

[image]

I am using the folder structure from release v3.0. My `device_repository.json` file is in the same folder as my `volumetric_capture.exe`. The relative path to my calibration images is `.\data\Calibrations\20-04-09-16-33-26\`. Are these the expected directory locations?

zuru, 2020-04-09:
Try moving it to a path without empty spaces (i assume it is in C:\Program Files currently).

mhudnell, 2020-04-09:
thanks for the response, that solved some of the errors. I still get the following: 
[image]

For the first error, I have built my device_repository.json and it is placed alongside volumetric_capture.exe, is this what it is looking for?

And for the second error, I'm trying to calibrate using only 2 out of the 4 cameras at the moment, so there are only 2 input depth images, is this supported?

vladsterz, 2020-04-09:
Hello @mhudnell ,
you have to place device_repository.json aswel under **Data\Executables\Calibration\multisensor_calibration\Resources\data** , I think we missed this step in our wiki. :/
At this current state calibration of N > 2 devices is supported.

mhudnell, 2020-04-09:
@vladsterz , thanks

I did not have the directory **multisensor_calibration** in my v3.0 release, and creating the full path and placing it there did not work. But you lead me in the right path, I placed it in **Data\Executables\Calibration** and it worked.

closed 2020-04-09 by mhudnell

vladsterz, 2020-04-09:
Oh you are on our v3.0 release. Note that this release uses our older calibration method, which works strictly for N = 4 devices.

mhudnell, 2020-04-09:
Yeah, in the release section it says Intel 415 isn't compatible with your v4.0 release so I went with v3. We may upgrade to Azure Kinects, would be nice to have more than 4 cameras.

vladsterz, 2020-04-09:
Our initial plan for Realease v4.0 was to support both device types simultaneously, using the new calibration method, which we achieved but in an unstable version, not ready for release. Under the pressure of CoViD19 situation, we pushed for a release that at least supports the new device type (kinect Azure), before quarantine was established (we work from home). A release supporting both types of devices, as well as an arbitrary number of sensors is definitely going to happen, but given the current situation I am unable to tell when :/ .