VCL3D/VolumetricCapture #20: Getting camera extrinsics

mhudnell, 2020-04-24:
I'm trying to load in the camera extrinsics that are given by the `volumetric_capture.exe` calibration (using v3.0). I found the .extrinsics files located in `volumetric_capture\Data\Calibrations\20-04-14-15-08-05\`. However I noticed the extrinsics in the file are slightly different from what is output in the `volumetric_capture.exe` console.

Example console extrinsics:
[image]

Example `d4150.extrinsics` file contents:
```
0.0124787 -0.999791 0.016157
-0.23122 0.0128353 0.972817
-0.972821 -0.0158753 -0.231011
-71.0407 -1803.81 494.83
```

Which is correct / what is the intended way for me to load camera extrinsics?
I'm using `multiview-player.exe` to generate calibrated clouds with `multiview-player.exe -f d4150.cdv --pcloud --is_pcloud_calibrated`, how does it know where to find the extrinsics, is it stored in the .cdv?

zokin, 2020-04-24:
The extrinsics pose of each viewpoint is indeed stored in the .cdv.

Should you want to use the *.extrinsics text file:
- It starts with the rotation matrix in row major format
- Then the translation parts follows in millimeters

The pose reported in the console is after conversion to meters, and flipping the `y-z` axis for correct upright rendering (given that the structure has `z-up`, while we want to render according to the sensor's `y-up`)

closed 2020-04-24 by zokin

mhudnell, 2020-04-24:
thanks! that cleared it up

mhudnell, 2020-04-30:
@zokin hi, apologies for maybe a simple / obvious question, but are the `*.extrinsic files` relative to the **depth camera**? If I want to get the pointcloud in the coordinates of the **color camera**, do I need to apply the "Color Depth Rotation" and "Color Depth Translation" transformations from the `device_repository.json`?

zokin, 2020-04-30:
The `*.extrinsics` files are transform the depth-derived points from the local coordinate system of each sensor to the common/global one, as defined by the calibration structure.

The _calibrated_ point cloud is thus defined on the global coordinate system. In order to transform it to the local _sensor_ coordinate system, you need to apply the inverse transformation (of the matrix as defined in the `*.extrinsics` files). 

As you correctly assumed, **this is at the original of the depth camera**, as the calibration process only uses the depth maps, and no color information. 

After you  transform it to the depth origin, you need to apply the depth to color transformation, in order to transform the pcloud at the color camera's original / coordinate system.

Hope that helps, also take into account that if you are using the previous release, the was an error in the keys, the transform is actually the depth to color rotation/translation, and not the color to depth ones as indicated by the keys.

mhudnell, 2020-05-01:
thanks you, that was very helpful, I managed to transform it correctly to color cam's coordinate system.