VCL3D/VolumetricCapture #21: Device_repository.exe cannot find device_respository.json

tylergibson, 2020-04-27:
[image]

Have spent an hour trying every combination imaginable to provide it a path.  No matter what is provided, a FATAL error results.  Simply copied the default device_repository.json to this directory.

zokin, 2020-04-27:
Hi Tyler, can you please attach the file (`device_repository.json`)?

xudi1227, 2020-05-01:
Hi @zokin  I also got the same problem using Azure Kinect. And here is my json file
[device_repository.zip](https://github.com/VCL3D/VolumetricCapture/files/4562005/device_repository.zip)

[image]

Basically I changed the Serial and Firmware to my own kinect specs as below.  
[image]

But still I can't add or list any Kinect using device_repository.exe. Not sure if I made any mistakes here.please let me know if someone solved this.

xudi1227, 2020-05-01:
Btw here is my command window
[image]

zokin, 2020-05-01:
Hi Johnny,

it seems that you are missing the `-c 0` or `--cam_type 0` argument when calling the `device_repository.exe` tool.

As the help also indicates, these are needed to disambiguate between the sensors.

The documentation about this is missing, but will be added/improved the coming days as well.

Thanks for your time and help in pinpointing issues.

tofis, 2020-05-01:
Hello all,

Another important missing point of the the [K4A alpha version](https://github.com/VCL3D/VolumetricCapture/releases/tag/4.0.1) is related to the new shaders for the K4A point cloud visualization which causes a crash. Please, find attached the missing shaders and place them in the "Resources" folder.

Sorry for the inconvenience but this release was not tested enough due to the sudden lock-down, as we have already said. Since we are coming back probably next week, we are going to fix the identified issues asap.

Thank you for your time. 
  
[depth_cloud_k4a_shaders.zip](https://github.com/VCL3D/VolumetricCapture/files/4564762/depth_cloud_k4a_shaders.zip)

tylergibson, 2020-05-02:
> Hi Johnny,
>
> it seems that you are missing the `-c 0` or `--cam_type 0` argument when calling the `device_repository.exe` tool.
>
> As the help also indicates, these are needed to disambiguate between [...]

This isn't possible, because as @xudi1227 also points out, the device_repository.exe tool always fails due to not locating the json file.  So we can't run it.  I also attempted to manually add my Kinects to the json file as @xudi1227 did, with no success.

Having a source release would make it much easier for us to debug and provide improvements ;)

tofis, 2020-05-02:
For the ones who use the new K4A alpha version, the recommended steps with respect to the correct usage of device_repository tool are given below:

- Assuming that there is no K4A device appropriately added in the "device_repository.json", delete any existing "device_repository.json" you copied in the root folder of the device_repository tool from other version or other folder, in order to make it clean from scratch.
- Place in the root folder of the device_repository tool the "depthengine_2_0.dll" which is missing (you can copy it from "remote_eye\bin" folder).
- Connect the first K4A device on a USB3 port (ensure that the device has been detected and runs properly, for instance using the MS Official K4A viewer).
- Run the "device_repository.exe" in the cmd using these arguments `device_repository.exe --add {device_name_prefer} --cam_type 0`.
- After that you are going to see the log you see in the screenshot below, while a new "device_repository.json" is going to be created. 
[image]
- Then, running  `device_repository.exe --list`, you are going to see your device listed if everything has been completed correctly.
[image]
I hope that it will work for you. As said, we hope at the end of the next week to have an actual release with documentation.

zuru, 2020-10-12:
Updated [documentation](https://vcl3d.github.io/VolumetricCapture/docs/configure/#creating-a-device-repository) with respect to the device repository is now available.

closed 2020-10-12 by zuru