VCL3D/VolumetricCapture #22: Azure Kinect mis-identified as Intel camera

tylergibson, 2020-04-27:
Camera appears in volumetric_capture as a camera:
[image]
but when selecting "Connect", the application crashes and the remote_eye.exe crashes as well with the following:
[image]

It appears I need to manually specify that the camera is an Azure Kinect, but I cannot do this due to #21

zokin, 2020-04-27:
Can you please verify that these (central volumetric capture and remote eye) are not running on the same machine?

Currently this is not supported due to socket port conflicts.

tylergibson, 2020-05-02:
Ah they were running on the same machine.  I will attempt on separate machines, although all the rabbitmq bits seem to work just fine.

zuru, 2020-05-03:
This is due to ptp sync needing UDP networking rather than the broker (RMQ) which handles the rest of the communication with the sensors.
Please let me know whether the issue persists or if  this is now resolved in order to close it.

zuru, 2020-10-12:
This has been fixed in the latest release, there is no port conflict and the same workstation can accommodate the `volcap` application as well as an `eye`.

closed 2020-10-12 by zuru