VCL3D/VolumetricCapture #23: Are Intel NUCs a hard requirement?

sergiobd, 2020-05-18:
I do not have Intel NUC's at the moment, but I can configure some PCs that were previously used in a render-farm in order to test VCL3D. 
Are Intel NUCs a hard requirement? (e.g. I tried to install CUDA 8.0 on my desktop with GeForce RTX 2060 and apparently CUDA 8.0 is too old for it).

Thanks!

LeoSarog, 2020-05-18:
Hi @sergiobd , 
Regarding your questions:

- The Intel NUCs are not essential, only recommended because our software changes the led color of the NUC based on the state (recording, calibrating etc.). They act also as indicative of the recommended hardware specifications for the current software.
- Releases 1.0, 2.0 and 3.0 require CUDA 8.0. Our latest releases (4.0+) do not require any CUDA version. Thanks to your comment, we already updated our wiki section on that.

zokin, 2020-05-18:
Indeed, NUCs are not necessary, but they offer a portable and easy to setup solution.

Take into account that without high quality USB 3.0 extensions you might have trouble setting it up with desktops as the sensor cables are short.

In any case, with the desktop setup you can either manually fire up the remote eyes or install the service and have them automatically start and connect.

closed 2020-05-18 by zokin

sergiobd, 2020-05-19:
Thank you for your very quick answers. I agree that Intel NUCs are very portable solution. I will probably upgrade the system to NUCs after the initial tests.