VCL3D/VolumetricCapture #25: Fail to stream with Kinect Azure

sergiobd, 2020-05-21:
Hi,

I am able to succesfully connect an Azure Kinect. Once I click "Connect" in the "Available Devices" widget, a snapshot of the scene appears both in the 3D space and in the color preview. But it seems to work only for a couple of frames (kinect light turns on and then turns off again). The console shows no errors.

I've tried with 2 different network switches. Any clues on what could be wrong? 

This is my remote_eye.log file:

```
2020-05-21 12:37:57.664 INFO  [6676] [main@66] Logging to: C:\Capturer\bin\remote_eye.log
2020-05-21 12:37:58.103 INFO  [6676] [main@73] Connected K4A device /w serial no: 000135500112
2020-05-21 12:37:58.103 INFO  [6676] [main@78] Connecting to RabbitMQ broker: amqp://volumetric:capture@192.168.0.121:5672 # Handshake
2020-05-21 12:37:59.109 INFO  [6676] [remote::handshake::say_hello@50] Sent hello from 000135500112
2020-05-21 12:37:59.109 INFO  [6676] [main@221] Waiting for welcome (press 'q' to quit) ...
2020-05-21 12:38:25.187 DEBUG [4276] [main::<lambda_19922d6ce101d1688e5e6b4773c2ac7d>::operator ()@104] Received Welcome:
{ msg = Welcome , serial = 000135500112 /w stream channel = Kinect01_c1280_x_720_d320_x_288_stream /w params channel = Kinect01_params /w workflow channel = Kinect01_workflow /w info channel = Kinect01_info /w preset = accuracy-100um /w sync type = Default
}
2020-05-21 12:38:25.187 INFO  [4276] [main::<lambda_19922d6ce101d1688e5e6b4773c2ac7d>::operator ()@116] Creating networking components...
2020-05-21 12:38:25.188 INFO  [4276] [main::<lambda_19922d6ce101d1688e5e6b4773c2ac7d>::operator ()@124] Streaming channel opened.
2020-05-21 12:38:25.193 ERROR [4276] [local::led_controller::led_controller@59] NUC Led controller app is not available @ C:\Capturer\bin\intel_nuc_led_utils.exe.
2020-05-21 12:38:25.193 INFO  [4276] [main::<lambda_19922d6ce101d1688e5e6b4773c2ac7d>::operator ()@134] Command channel opened.
2020-05-21 12:38:25.193 INFO  [4276] [main::<lambda_19922d6ce101d1688e5e6b4773c2ac7d>::operator ()@154] Initializing syncer on C:\Capturer\bin\PTPSync_slave.exe ...
2020-05-21 12:38:25.193 INFO  [4276] [main::<lambda_19922d6ce101d1688e5e6b4773c2ac7d>::operator ()@160] Syncer initialized.
2020-05-21 12:38:25.193 INFO  [4276] [main::<lambda_19922d6ce101d1688e5e6b4773c2ac7d>::operator ()@163] Initializing device w Color-Depth streams.
2020-05-21 12:38:25.193 INFO  [4276] [main::<lambda_19922d6ce101d1688e5e6b4773c2ac7d>::operator ()@170] Loading preset <accuracy-100um> from C:\Capturer\bin\Resources\accuracy-100um.json
2020-05-21 12:38:25.234 INFO  [4276] [main::<lambda_19922d6ce101d1688e5e6b4773c2ac7d>::operator ()@175] Acquiring device parameters...
2020-05-21 12:38:25.284 INFO  [4276] [main::<lambda_19922d6ce101d1688e5e6b4773c2ac7d>::operator ()@185] Device parameters initialized from configuration file: C:\Capturer\bin\Resources\accuracy-100um.json
2020-05-21 12:38:25.355 DEBUG [18880] [main::<lambda_19922d6ce101d1688e5e6b4773c2ac7d>::()::<lambda_ee3d8ba1313f35c7f4e3ff6a4d2d9a7e>::operator ()@145] Received Workflow Update Command:
{
id = 0
time = 66575301200362
action = 2
dev param = 1
value = 000001FA3254EBD0
}
2020-05-21 12:38:25.409 INFO  [4276] [main::<lambda_19922d6ce101d1688e5e6b4773c2ac7d>::operator ()@187] Device started.
2020-05-21 12:38:25.410 INFO  [4276] [main::<lambda_19922d6ce101d1688e5e6b4773c2ac7d>::operator ()@208] Device parameters session created.
2020-05-21 12:38:25.410 INFO  [4276] [main::<lambda_19922d6ce101d1688e5e6b4773c2ac7d>::operator ()@213] Encoding /w compression params:

ENTROPY: { Type = 3 , Level = 4 , Shuffle = 1 }
2020-05-21 12:38:25.410 INFO  [4276] [main::<lambda_19922d6ce101d1688e5e6b4773c2ac7d>::operator ()@217] Streaming pipeline ready.
2020-05-21 12:38:25.504 DEBUG [18880] [main::<lambda_19922d6ce101d1688e5e6b4773c2ac7d>::()::<lambda_ee3d8ba1313f35c7f4e3ff6a4d2d9a7e>::operator ()@145] Received Workflow Update Command:
{
id = 1
time = 66575407463722
action = 2
dev param = 6
value = 000001FA32E58F90
}
2020-05-21 12:38:25.507 DEBUG [18880] [main::<lambda_19922d6ce101d1688e5e6b4773c2ac7d>::()::<lambda_ee3d8ba1313f35c7f4e3ff6a4d2d9a7e>::operator ()@145] Received Workflow Update Command:
{
id = 1
time = 66575407463722
action = 3
dev param = 6
value = 000001FA32E590B0
}
2020-05-21 12:38:25.656 DEBUG [18880] [main::<lambda_19922d6ce101d1688e5e6b4773c2ac7d>::()::<lambda_ee3d8ba1313f35c7f4e3ff6a4d2d9a7e>::operator ()@145] Received Workflow Update Command:
{
id = 2
time = 66575513731315
action = 2
dev param = 7
value = 000001FA32E59140
}
2020-05-21 12:38:25.806 DEBUG [18880] [main::<lambda_19922d6ce101d1688e5e6b4773c2ac7d>::()::<lambda_ee3d8ba1313f35c7f4e3ff6a4d2d9a7e>::operator ()@145] Received Workflow Update Command:
{
id = 3
time = 66575619986279
action = 2
dev param = 8
value = 000001FA32E58FC0
}
2020-05-21 12:38:25.958 DEBUG [18880] [main::<lambda_19922d6ce101d1688e5e6b4773c2ac7d>::()::<lambda_ee3d8ba1313f35c7f4e3ff6a4d2d9a7e>::operator ()@145] Received Workflow Update Command:
{
id = 4
time = 66575726235668
action = 4
dev param = 7
value = 000001FA32E59550
}

```

zuru, 2020-05-21:
Hi Sergio,

this is a bug currently, we will investigate it and fix it as it manifested only in the K4A release and we have also confirmed it.

For now, as a workaround, after the first connection (when everything freezes), please refresh all connected devices and then reconnect them. Please pay attention that in case you get persistent black streaming views, then you need to close and restart the application unfortunately.

Sorry for the inconvenience, once we fix and update it, we will notify you.

jessekirbs, 2020-08-01:
I am experiencing the same issue. It sometimes resolves when clicking 'Refresh' and then 'Connect' for the devices. Any updates on this fix? Thank you!

zokin, 2020-08-01:
We've identified the source of this bug, it will get fixed in the next actual release alongside many other issues.
However, due to overall unfortunate situations that resulted in hectic priority rearrangements, it is not possible to say when that will be.
For now, simply refresh and re-connect, and if that doesnt work, please restart the application.

jessekirbs, 2020-08-01:
@zokin Thanks for your quick reply! No problem at all - refreshing and reconnecting works. Thanks, Nikolaos!

mellinger, 2020-10-01:
Hey! Thanks for this tremendous software, I'm really excited to get it working and explore workflows with the output. 

I believe I am also getting this error, except in my situation the Kinect Azure LED stays on. Otherwise, the issue is identical (workstation gets a couple of frames, then nothing, no errors in either the NUC or Workstation logs). 

Attached is a screenshot of my workstation, the workstation log, and the NUC log. 

[image]

No amount of refreshing or restarting the application seems to address it. 

Let me know if this is the same issue or if something is else awry. Excited for the new release!

[remote_eye_service.log](https://github.com/VCL3D/VolumetricCapture/files/5309511/remote_eye_service.log)
[workstation-log.txt](https://github.com/VCL3D/VolumetricCapture/files/5309512/workstation-log.txt)

zuru, 2020-10-12:
This is now fixed in the latest release.

closed 2020-10-12 by zuru