VCL3D/VolumetricCapture #29: Calibration Fails

jessekirbs, 2020-08-01:
I am following the Wiki and turning both iteration options in _Calibration>Configure_ up to the max and then going to _Calibration>Capture_. It seems to successfully capture the images. I then go to _Calibration>Calibrate_ and it runs for a little bit before giving me the following error:

```
Device: JUP01, correspondences: 1, error: 1.5940710465918552e-32
Traceback (most recent call last):
File "Data\Executables\Calibration\multisensor_calibration\calibration.py", line 99, in <module> main(sys.argv)
File "Data\Executables\Calibration\multisensor_calibration\calibration.py", line 47, in main "--save_pointclouds", "True"
File "C:\VCLVolumetric\Release_4_0_2\volumetric_capture\Data\Executables\Calibration\multisensor_calibration\src\structurenet_calibration.py", line 138, in main predicted_centers = np.stack([pcloud[np.transpose(label_map) == id + 1,:].mean(axis = 0) for id in valid_ids])
File "C:\Users\jesse\anaconda3\lib\site-packages\numpy\core\shape_base.py", line 412, in stack raise ValueError('need at least one array to stack')
ValueError: need at least one array to stack
```
I've made sure to update the calibration tool as explained in the Wiki by running 'python.exe install.py'.

I tried running calibration last night as well, and it seemed to work (no error messages) but when I clicked Load Last Config nothing happened and the point clouds didn't move.

One thing to note (this may be a separate bug so I'll open as a new issue as well) is that the pointclouds displayed are very shallow. I've increased the boundaries to the maximum but it still only captures up to about six feet away, which is much less than the Kinect is capable of. I've tested in the k4aviewer app and it can capture the entire distance of my room. Due to this issue, it seems like it's not getting much info from the calibration boxes.

Any help would be greatly appreciated. Thank you!

**Workstation PC:**
Windows 10 Home x64 v. 1909
Intel i9-9900K @ 3.60 GHz
32gb RAM
NVIDIA GTX 2060

**Capture PCs:**
Windows 10 Pro
(3) Intel NUC i7 (NUC7i7BNH)
8gb RAM

(3) Kinects for Azure
Kinect for Azure SDK 1.4.1
Firmware v1.6.110079014

**Network Switch:**
Netgear Unmanaged 8-port Gigabit Switch
225gb SSD

jessekirbs, 2020-08-07:
This issue seems to have been caused by only using two Kinects. After integrating a third, the calibration worked. There was, however, still an issue with a certain configuration of the three sensors. 

[image]
(Setup 1)

When using Setup 1, the calibration completes successfully, however Kinect 3 is not aligned properly. Kinect 1 & 2 are aligned perfectly. I tried several calibrations as well as moving the calibration boxes to different angles but I had the same result. 

[image]
(Setup 2)

When using Setup 2, calibration was great. 

Are there certain configurations of sensors that don't work? Can you offer any tips to possibly get Setup 1 working? Thanks!

zuru, 2020-10-12:
Yes, it is harder to calibrate just `3` sensors, unfortunately. 
Best performance is usually achieved when the sensors look at `8` of the boxes' sides.
However, with some trial-n-error, it is possible to calibrate just `3` sensors, see the example [here](https://vcl3d.github.io/VolumetricCapture/docs/acquisition/#issue29).