VCL3D/VolumetricCapture #30: Bad/Inconsistent Framerate with Three Kinects

jessekirbs, 2020-08-01:
Everything works well when two Kinects are activated and framerates are @ 30fps. When I activate the third Kinect, two of the streams are fine but the third lags badly and the framerate jumps around quite a bit. [Here is a video of the issue](https://www.dropbox.com/s/dtlc0smuztnzqhn/FPSIssue.mkv?dl=0). Let me know if I can provide anything else to help diagnose this. 

**Workstation PC:**
Windows 10 Home x64 v. 1909
Intel i9-9900K @ 3.60 GHz
32gb RAM
NVIDIA GTX 2060

**Capture PCs:**
Windows 10 Pro
(3) Intel NUC i7 (NUC7i7BNH)
8gb RAM

(3) Kinects for Azure
Kinect for Azure SDK 1.4.1
Firmware v1.6.110079014

**Network Switch:**
Netgear Unmanaged 8-port Gigabit Switch
225gb SSD

zokin, 2020-08-01:
This is hard to answer but some things that may help:

- Just to be sure check that no network/bandwidth issues are the cause (i.e. 100Mbit cable instead of 1Gbit one)
- Use lower resolutions are we are currently using the K4A jpeg compression which uses a higher quality, and thus higher BW. The same goes for the depth resolution as we are using lossless coding there.
- The constant 15 FPS on the problematic device can be the cause of automatic exposure, try to set it to manual.
- Check that specific NUC for any anti-virus popping up, etc. wasting computational resources (use remote desktop, mouseover the name on the available devices and you'see the IP, you can even ctrl+c copy it when moused over).
- Flip the sensor to another NUC to see if the problem is because of the sensor or the NUC.

I'm pretty sure the list is longer but cannot remember everything at the moment :)

jessekirbs, 2020-08-01:
@zokin Fantastic - I appreciate the reply. I'll try all these out and report back.

jessekirbs, 2020-08-07:
@zokin I had the chance to do some experimenting again today and after several times refreshing, reconnecting, and restarting the VCL software, eventually the three Kinects started hitting 30fps. I didn't make any changes to the hardware or software besides refreshing and restarting four or five times. I also went through your suggestions:

**Just to be sure check that no network/bandwidth issues are the cause (i.e. 100Mbit cable instead of 1Gbit one)**

The network switch is an unmanaged 1gbps unit and I'm running 1gbit cables.

**Use lower resolutions are we are currently using the K4A jpeg compression which uses a higher quality, and thus higher BW. The same goes for the depth resolution as we are using lossless coding there.**

After restarting and hitting 30fps, I tried many different settings and the 30fps was maintained. 

**The constant 15 FPS on the problematic device can be the cause of automatic exposure, try to set it to manual.**

I don't see this setting anywhere. Am I missing something or is this only available for the Intel RealSense? Please let me know where to find this setting so I can test it out. 

**Check that specific NUC for any anti-virus popping up, etc. wasting computational resources (use remote desktop, mouseover the name on the available devices and you'see the IP, you can even ctrl+c copy it when moused over).**

The NUC is a fresh Windows install with no additional software. They seem to be working fine. 

**Flip the sensor to another NUC to see if the problem is because of the sensor or the NUC.**

The issue is not confined to the sensor. When the low framerate occurs, it's always whichever sensor is added last when selecting 'Connect All'. Two sensors connected is always a solid 30fps, and when adding a third the issue occurs - but not always. Again, after multiple restarts it seems to eventually work on occasion. 

Are there any methods you can think of to diagnose why this happens intermittently? I'm happy to do more experimenting, just let me know.

zuru, 2020-10-12:
Hi Jesse, the first time crash issue should be fixed now in [5.0.0]().

With respect to multi-sensor streaming, with the appropriate hardware and stream profile settings, it is possible to stream and record `6-8` sensors in real-time as seen in the [examples](https://vcl3d.github.io/VolumetricCapture/docs/acquisition/#issue30) section of the docs.

closed 2020-10-12 by zuru