VCL3D/VolumetricCapture #31: Crash When Changing Depth Units Value

jessekirbs, 2020-08-05:
The default depth units setting is 1000 and this is too shallow for my capture space. When I try to change the value on this slider, the program crashes. I've tried restarting everything and attempting multiple times. Just to be sure, changing this value increases the depth of what is being captured, correct? If not, please let me know the correct way to achieve this. 

I believe the capture depth being too shallow is also causing the issue I'm seeing with calibration (https://github.com/VCL3D/VolumetricCapture/issues/29). I'm not able to capture enough information from the boxes due to the capture space being too shallow.

Please let me know if I can help by providing more info or logs of any kind. Thank you!

**Workstation PC:**
Windows 10 Home x64 v. 1909
Intel i9-9900K @ 3.60 GHz
32gb RAM
NVIDIA GTX 2060

**Capture PCs:**
Windows 10 Pro
(3) Intel NUC i7 (NUC7i7BNH)
8gb RAM
225gb SSD

(3) Kinects for Azure
Kinect for Azure SDK 1.4.1
Firmware v1.6.110079014

**Network Switch:**
Netgear Unmanaged 8-port Gigabit Switch

tofis, 2020-08-06:
Hello @jessekirbs,

Thank you for opening this issue, we will check the crash you mention. However, from your description, I think that you are referring to the depth threshold with respect to the maximum allowed depth distance. What is the distance of the sensors from the center of the capturing space? Can you please share a screenshot of your setup and another from the captured point-cloud?

jessekirbs, 2020-08-06:
@tofis I assumed the 'Depth Units' slider was the way to increase/decrease the maximum depth distance - but yes, I am looking for a depth threshold to increase the maximum allowed depth distance. Is there a way to adjust this?

The distance to the center of my capture space is about 6 ft. (1.8m). I would like to have the option to be able to capture from even further away too, if possible. 

[This video](https://www.dropbox.com/s/7fnmq2kd7ass0f8/CalibrationIssue.mkv?dl=0) from my calibration issue post should give you a good idea of my setup and pointclouds. I have three Kinects and am purchasing a fourth, but I want to make sure everything is working with two first. Thanks!

zuru, 2020-08-06:
Hi Jesse,

the depth units slider is not working with the K4A devices, it is mainly there to support that specification for the RS2 devices, where it is directly related with the maximum depth captured (it will be fixed on that future release).

The maximum depth captured is always what the K4A device offers, it is only cropped for visualization purposes (you can verify this by checking the calibration captured depth images using Irfraview for example).

Nonetheless, our calibration is only trained up to a certain range (see https://github.com/VCL3D/StructureNet) and is currently focused at closer range captures. Also, the minimum viable calibrating setup should be with 3 devices at the moment.

Regarding the calibration error itself, please send us the respective folder in `./Calibrations`, although, due to summer break, we will, unfortunately, be able to investigate it in 2 weeks' time.

jessekirbs, 2020-08-06:
@zuru Ah, I see! Thanks for letting me know. The calibration issue seems to have been caused by only using two sensors. I added the third and calibration worked! I'll update that issue. Appreciate the help!

zuru, 2020-10-12:
The depth units are now fixed for RS2 and K4A, as indicated in the [5.0.0](https://github.com/VCL3D/VolumetricCapture/releases/tag/5.0.0) release notes.

Calibration works for `>=3` sensors, but for just `3` the appropriate placement is trickier to achieve, as also mentioned [here](https://vcl3d.github.io/VolumetricCapture/docs/acquisition/#issue29). It is mostly an issue of re-positioning the sensors for the model to achieve a good enough segmentation (these can be inspected in `./Calibrations/LATEST_CAPTURES`.

closed 2020-10-12 by zuru