VCL3D/VolumetricCapture #33: Cannot Dump Colored Pointcloud from Multiview Player GUI

jessekirbs, 2020-08-07:
After opening the recorded sequences in the multiview player GUI and selecting the 'Dump' options, I don't see a way to dump the colored pointclouds - only the color and depth image sequences. Is this possible through the GUI?

I've also attempted to dump the colored pointcloud from the command line tool, but I am unable to figure out the correct syntax. Can you please provide an example syntax for how to dump colored, calibrated pointclouds from three or more recorded .cdv files? Thank you!

jessekirbs, 2020-08-17:
Hey - are there any updates on this or when you might be able to address it? Thank you!

vladsterz, 2020-08-18:
Hello @jessekirbs ,
we've disabled this feature in our latest releases due to some changes we've made after the Kinect Azure release. We plan to re-enable this feature in our upcoming release, so stay tuned.
Vlad

jessekirbs, 2020-08-18:
@vladsterz Thanks for the reply. Do you have any idea of the timeframe for the next release? In the meantime, do you know of any other software I could use to convert the color and depth images dumped from this to a pointcloud or mesh? Thanks, Vlad.

vladsterz, 2020-08-18:
@jessekirbs it is hard to estimate the date for the next release, but it should not be too far away. We already have detected core issues and plan to start resolving them ASAP. Please be patient and stay tuned :)
Vlad

jessekirbs, 2020-08-18:
@vladsterz I'm looking forward to it - you guys are doing some great work here.

zuru, 2020-10-12:
Closing as this is now supported in the [5.0.0](https://github.com/VCL3D/VolumetricCapture/releases/tag/5.0.0) release, see [here](https://vcl3d.github.io/VolumetricCapture/docs/acquisition/#issue33).

closed 2020-10-12 by zuru