VCL3D/VolumetricCapture #37: Add additional step to installing remote eye to documentation

mellinger, 2020-10-11:
In my experience, you have to run the Remote Eye app as an administrator and OK the security dialog that appears before running the install script, otherwise the install script will fail. Might be nice to note this in the documentation.

ankarako, 2020-10-12:
Hi @mellinger ,

Actually, there is no install script for the `remote_eye` application. Is there a chance that you mean the **Monitor** Service (i.e. the `remote_eye_service` Windows Service? If so, we mention in the [documentation](https://vcl3d.github.io/VolumetricCapture/docs/software/#service--eye) that you need to run the installation script with administrator rights.

Best,
Antonis

mellinger, 2020-10-12:
Hi! In my experience, you have to run remote_eye_service.exe as an administrator and OK the security warning once BEFORE you run the install script as an administrator, otherwise you get an error.

ankarako, 2020-10-13:
Hi @mellinger,

Is there a chance that you had already installed the `remote_eye_service` from the previous version? Then you'll have to uninstall it (by running the uninstall script with administrator rights) and then run the install script for the new version.
I'll make sure we add this in the documentation.

Best,
Antonis

mellinger, 2020-10-13:
On one, but not on the other 3.

ankarako, 2020-10-13:
Hi @mellinger,

If you want you can make a pull request and document it in the "Software Setup" section to further discuss it there. ;)

Best, 
Antonis

ankarako, 2020-10-13:
closing as the uninstall step has been documented and further steps will be discussed in the pull request

closed 2020-10-13 by ankarako