VCL3D/VolumetricCapture #38: Volcap not listing the devices from the device_repository.json

adri1177, 2020-11-20:
Hello,
I have installed everything in two computers, one as the main for the volcap app and another for the remote eye.

I have first configured the devices with the dev_repo.exe and it creates the file device_repository.json inside Resources succesfully.

I have tested it with the last release (5.0.0) and i am not getting any errors executing volcap.exe but no single device is listed on the app. I have tried this with the remote connected to the rabbitMQ but i understand that it should not be necessary.

I attach here the device_repository.json file that is generated.
Does anyone know what could be the issue?
Thanks in advance.

[device_repository.json.txt](https://github.com/VCL3D/VolumetricCapture/files/5573450/device_repository.json.txt)

zuru, 2020-11-20:
So, the issue is with the automatic connectivity through the service?

> I have tried this with the remote connected to the rabbitMQ but i understand that it should not be necessary.

This means that when you run it manually, it connects fine, right?

Also, the `.txt` extension on `device_repository.json` was added when uploading to GitHub?

adri1177, 2020-11-20:
The issue is with the volcap.exe that is not showing the devices that is loading from the device_repository.json

Does the loading of the devices is independent from the remote eye connection? I mean with a proper json the devices should be listed on the volcap.exe (just listing the devices)?

The .txt extension is because it was added to github.

Thank you.

zuru, 2020-11-20:
Volcap will only automatically show the available devices if the service is operational (see https://vcl3d.github.io/VolumetricCapture/docs/eyes/#automatic-sensor-connection).

Otherwise you need to manually run the eyes (see https://vcl3d.github.io/VolumetricCapture/docs/eyes/#manual-sensor-connection).

Let me know what happens when you manually run it. If possible please paste an image of the cmd line for both executables.

ankarako, 2020-11-20:
Hi @adri1177 ,

volcap should not display the devices in `device_repository.json` if they are not available in the network.

If I understood correctly, you successfully connected to a device manually:
>I have tried this with the remote connected to the rabbitMQ but i understand that it should not be necessary.

(i.e. by running `remote_eye.exe` via cli).

If the above is true, and you actually managed to connect to a device manually, then was the device name displayed properly on volcap? If yes, then your `device_repository.json` file should not have an issue.

On Windows, please follow the steps below: 
1. Go to "Control Panel -> Network & Sharing Center -> Change Adapter Settings"
2. Check how many network adapters exist on you PC. 
3. If there are more that one (1) network adapters on your system, (and especially if the rest network adapters are virtual -i.e. prefixed with "v" such as "vEthernet...") disable them (right click the adapter icon and select "Disable" - don't worry you can enable them again at any time, but they confuse volcap) 
4. Check again the automatic connection.

adri1177, 2020-11-20:
Hello @zuru @ankarako 
I was not able to make it run manualy till now. Seems like i have to run first the volcap.exe and then launch manually the remote_eye_k4a.

Seems like service is not working for us. No other network adapters are present on our systems.
I will update this after doing more test to check if i manage to make services work.
Thanks

ankarako, 2020-11-20:
> I was not able to make it run manualy till now. Seems like i have to run first the volcap.exe and then launch manually the remote_eye_k4a.

Yes, that is indeed the correct order of running things manually.

Please open the Windows "Task Manager" on the PC running `remote_eye` and then go to the "Services" tab. Try to find `remote_eye_service` and check if its status is "Running".
If the status is "Stopped" then right-click the service and click "Run".
If the status is immediately changes to "Stopped" then something is wrong with the service installation. Please check some troubleshooting advice [here](https://vcl3d.github.io/VolumetricCapture/docs/eyes/#troubleshooting).

closed 2020-12-11 by zuru