VCL3D/VolumetricCapture #39: An attempt was made to access a socket in a way forbidden by its access permissions

sumedhthesawant, 2020-11-23:
- At this point we have we have set up RabbitMQ, Setup the firewall with port TCP 5672(IN, OUT), TCP 15672(IN, OUT),UDP 11234(IN, OUT) & UDP 320,321. On master and Subordinate PCs
- We have downloaded Volcap Files 5.0.0 
- Configured all services in subordinate PC (Win Update, remote desktop,etc)
- Installed REMOTE_EYE_SERVICE on all PCs
- In RabbitMQ we have set up Volumetric/capture User/Password
- Device repository files has also been created for Microsoft Azure.

NOW, while starting the volcap.exe we are getting this error and we are not able to view our sensors.

```
Connecting to broker: { amqp://volumetric:capture@192.168.0.185:5672}
Exception: send_to: An attempt was made to access a socket in a way forbidden by its access permissions
```

We have tried opening the Volcap application with this command which has the custom IP found on remote_eye_k4a.exe `volcap.exe -b 195.251.117.104 -u Volumetric -p capture` Still no luck. Getting the same error

Any help would be highly appreciated. 
Thank You!

ankarako, 2020-11-23:
Hi @sumedhthesawant ,

This error possibly has to do with another application on your system that holds on a port that volcap tries to use. Could you open your terminal and run:

sumedhthesawant, 2020-11-23:
I checked in terminal for the port but no other application seem to be using it.

ankarako, 2020-11-23:
Could you temporarily disable the firewall of the PC running volcap, and check again please?

Thanks.

closed 2020-12-11 by zuru

annop584, 2021-03-19:
I have this issue too.
How could I fix this?

Connecting to broker: { amqp://volumetric:capture@192.168.1.168:5672}
Exception: send_to: An attempt was made to access a socket in a way forbidden by its access permission

[image]

zuru, 2021-03-19:
Please check the ports that need to be allowed by your firewall [here](https://vcl3d.github.io/VolumetricCapture/docs/software/).

ankarako, 2021-03-19:
@annop584 

Could you tell us a little more about your setup please?

annop584, 2021-03-19:
> Please check the ports that need to be allowed by your firewall [here](https://vcl3d.github.io/VolumetricCapture/docs/software/).

-I'm already did this just tcp 5672 port (In/Out)

zuru, 2021-03-19:
That is for the broker, other ports are required as well for synchronization (PTP) as well as indicated in the above [link](https://vcl3d.github.io/VolumetricCapture/docs/software/) and the original poster's description: `UDP 11234(IN, OUT) & UDP 320,321`.

Let us know if you still have the port crash after opening these as well.

annop584, 2021-03-19:
> @annop584 
> 
> 
> 
> Could you tell us a little more about your setup please?

Thank you very much for your reply

My installation step is

-download last release

-install RabbitMQ via chocolately, then I can open localhost:15672 to open RabbitMQ admin page to add  user (volumetric:capture)
(I can see RabbitMQ in services taskmanager)

-set firewall rule to open TCP port 5672 (In/Out)

-using dev_repo.exe to add device into repository (Kinect Azure)

-Open volcap.exe by double click, the volcap Window can launch but cannot see any device,
In console have error log “Connecting to broker: { amqp://volumetric:capture@192.168.1.168:5672}
Exception: send_to: An attempt was made to access a socket in a way forbidden by its access permission”

ankarako, 2021-03-19:
Did you install the the `remote_eye_service`? 
If not you can follow [these](https://vcl3d.github.io/VolumetricCapture/docs/software/#service--eye) instructions, in order to achieve automatic sensor connectivity.

However, if you don't mind manually connecting each device, you can skip it, but you need to [connect manually](https://vcl3d.github.io/VolumetricCapture/docs/eyes/#manual-sensor-connection) to each device.

You probably get this error because you didn't install the `remote_eye_service` (and as a consequence you didn't open the corresponding ports on your system for connecting to the service)

annop584, 2021-03-22:
Thank you @ankarako and @zuru for suggestion

Ok I will update my progress

1) install remote_eye_service by run as admin at file install_remote_eye_service.bat

2) set firewall rule for open UDP port 11234, 320, 321 (In | out) 

3) start volcap.exe again and the errors log still same

4) then I read this issue  https://github.com/VCL3D/VolumetricCapture/issues/42 
**because I just use only one local machine too**

5) I start a vocal.exe and open another console at Capture path  then run command
remote_eye_k4a.exe --address "mylaptopIP" --port 5672 --user volumetric --pass capture

6) Now error still same but I can see a camera from Kinect and can recording

figure 1: error still show on console
[image]

figure 2: but software can open a camera and recording
[image]

ankarako, 2021-03-22:
Hi @annop584 ,

We are very happy that you managed to make it through. Could you please check if following the steps described in the 3rd bullet [here](https://vcl3d.github.io/VolumetricCapture/docs/eyes/#troubleshooting) fixes the error message?