VCL3D/VolumetricCapture #41: Volumetric Capture without hubs?

zapaishchykova, 2020-12-29:
Hi! Is it possible to use the volcap.exe without clients (e.g. with direct usb connection to the computer)?

ankarako, 2020-12-29:
Hi @zapaishchykova,

Actually, with the latest version you should be able to do it (either manually, or via installing the  connection windows service on the main PC).

Best,
Antonis

zapaishchykova, 2020-12-29:
Hi @ankarako,
So what i did is i captured the device with dev_repo, which created me a .json, now I launch volcap and on the same computer i launch remote_eye. Is that a correct approach or I am missing something?

ankarako, 2020-12-29:
Yeah, your approach seems to be correct. You should open the `remote_k4a/rs2.exe' (depending on the device you are using), with the corresponding [cli arguments ](https://vcl3d.github.io/VolumetricCapture/docs/eyes/#manual-sensor-connection).

zapaishchykova, 2020-12-29:
Could you please tell me what arguments i should use? (i dont really get what does broker means and what config file should i pass? is it thedevice_repository.json? i have intel rs 435i)

ankarako, 2020-12-29:
OK, 

So the thing is that volumetric capture only supports Kinect 4 Azure and Intel Realsense D415 devices (see #40 ).
However here is an explanation of the `remote_k4a/rs2.exe` arguments:
* `--address`: The IP of the PC on which the RabbitMQ server (broker) runs on. (In your case this should be the local IP of the pc running volcap)
* `--port`: Is actually not needed (default is 5672), and it's the port for communicating with the RabbitMQ server.
* `--user`: should be "volumetric" (if you followed the RabbitMQ installation [instructions](https://vcl3d.github.io/VolumetricCapture/docs/software/#rabbitmq-setup)).
* `--pass`: should be "capture" (again if you followed the RabbitMQ installation [instructions](https://vcl3d.github.io/VolumetricCapture/docs/software/#rabbitmq-setup)).

So you should run: `remote_k4a/rs2.exe --address "your.rmq.i.p" --port 5672 --user volumetric --pass capture`.

zapaishchykova, 2020-12-30:
Oh that's unfortunate. Thank you for your help!

closed 2020-12-30 by zapaishchykova