VCL3D/VolumetricCapture #42: Trying the system without NUCs

kwansik, 2021-01-23:
Hi,

I have followed pretty everything in  https://vcl3d.github.io/VolumetricCapture/ including adding device repo, install rabbitmq and add user, etc, except 2 things

1. No NUCs, all devices(2 of Azure Kinect) are directly connected to host machine(localhost) vua USB3
2. No hardware sync cable between 2 Kinect

But when I starts volcap, there are no devices listed under "device".

So I was wondering if there are things I should do differently if I configure devices directly connected to localhost.
Let me know. 
Thanks.

ankarako, 2021-01-23:
Hi @kwansik,

Thanks for your interest in our project.
That setup, i.e. 2 devices connected on a single PC wasn't a use case and it isn't tested. I can think of two possible reasons that it isn't supported as well:

1. When the `remote_eye_service` is requested to spawn a `remote_k4a` process, it will check if the process is already open and if it is it will kill it and then spawn a new one. This means that the service will open one `remote_k4a` process but when it will be requested to open the second one, it will kill the first one.
You can stop the `remote_eye_service` (from the "Services" Tab in "Task Manger") and try to connect the `remote_ka4` application [manually](https://vcl3d.github.io/VolumetricCapture/docs/eyes/#manual-sensor-connection) , to see if this work around fixes this problem.

2. Another thing is that if there are 2 `remote_ka4` processes running, when they are requested to synchronize (via [nPTP](https://vcl3d.github.io/VolumetricCapture/docs/synchronization/#nptp), and [PTPd](https://vcl3d.github.io/VolumetricCapture/docs/synchronization/#ptpd)) both the processes will try to open the same port (on the same PC) to communicate with `volcap`. This will have undefined behavior and probably crash one of the processes.

Best,
Antonis

kwansik, 2021-01-24:
Thanks for a quick reply.

I could run remote_eye_k4a for a single device but we have problem running remote_eye_service
and thus could not connect to valcap, I think:

c:\Windows\Microsoft.NET\Framework\v4.0.30319\InstallUtil.exe remote_eye_service.exe
Microsoft (R) .NET Framework Installation utility Version 4.8.3752.0
Copyright (C) Microsoft Corporation.  All rights reserved.

Exception occurred while initializing the installation:
System.IO.FileLoadException: Could not load file or assembly 'file:///C:\Users\3080\VCL3D\release_5.0.0\remote\remote_eye_service\remote_eye_service.exe' or one of its dependencies. Operation is not supported. (Exception from HRESULT: 0x80131515).

ankarako, 2021-01-24:
You shouldn't _run_ `remote_eye_service` manually by double-clicking on it etc. You should [install](https://vcl3d.github.io/VolumetricCapture/docs/software/#service--eye) it on the system, and then it runs on the background. 

If you followed the instructions and the service is installed, in order to connect manually to `volcap` (i.e. without using `remote_eye_service`'s automatic connectivity) you should `Stop` the service (i.e. open _Task Manager_ and go to _Services_ tab, then find _remote_eye_service_, right-click it and select _Stop_) and then follow the steps below:
1. Open `volcap`
2. Run `remote_k4a.exe` via the _Command Line_ as described [here](https://vcl3d.github.io/VolumetricCapture/docs/eyes/#manual-sensor-connection). Again, If you followed the instructions the command line arguments that will be needed will be only the `--address` argument, which is the IP of the pc that RabbitMQ runs on (i.e. the IP of your PC).
3. The device should appear on `volcap` and hit _Connect_ to connect with it.

Best,
Antonis

kwansik, 2021-01-24:
Same results. It fails to install remote_eye_service.

ankarako, 2021-01-24:
In my previous message I mentioned _If you followed the instructions and the service is installed_, because in your original message you said that you followed our instructions. So If you tried to install it again the installation process will fail.

Is there a case that the `remote_eye_service` was already installed? 

However, if you need to run `volcap` on a single PC (i.e. with the device connected on that same PC) you don't need the `remote_eye_service` installed.

You can try to connect manually without the service installed or running.

kwansik, 2021-01-24:
I tried remote_eye_service because manual connection did not work.
I do not see any device listed on volcap.
Following is the outputs just in case...
Thanks.

--------------------------------------------------------
C:\Capturer\bin>remote_eye_k4a.exe --address 192.168.186.113
Logging to: C:\Capturer\bin
Connected K4A device /w serial no: 000934503412
Connecting to RabbitMQ broker: amqp://volumetric:capture@192.168.186.113:5672 # Handshake
Sent hello from 000934503412
Waiting for welcome (press 'q' to quit) ...

---------------------------------------------------------------------<tried with both "Volumetric" and "volumetric">---------------

C:\Users\3080\VCL3D\release_5.0.0\volcap>volcap.exe -l 192.168.186.113 -b 192.168.186.113 -u Volumetric -p capture -v
Connecting to broker: { amqp://Volumetric:capture@192.168.186.113:5672}

GL_VENDOR    : NVIDIA Corporation
GL_RENDERER  : GeForce RTX 3080/PCIe/SSE2
GL_VERSION   : 3.2.0 NVIDIA 457.30
GLSL_VERSION : 1.50 NVIDIA via Cg compiler

DDRenderInterfaceCoreGL initializing ...
> DDRenderInterfaceCoreGL::setupShaderPrograms()
> DDRenderInterfaceCoreGL::setupVertexBuffers()
DDRenderInterfaceCoreGL ready!

Connecting to broker: { amqp://Volumetric:capture@192.168.186.113:5672}
refresh ips: 192.168.186.113 Volumetric capture 5672 C:\Capturer\bin remote_eye.log default.json
Connecting to broker: { amqp://Volumetric:capture@192.168.186.113:5672}
Connecting to broker: { amqp://Volumetric:capture@192.168.186.113:5672}
refresh ips: 192.168.186.113 Volumetric capture 5672 C:\Capturer\bin remote_eye.log default.json
Connecting to broker: { amqp://Volumetric:capture@192.168.186.113:5672}

ankarako, 2021-01-24:
Just in case, you should first open `volcap` and then run `remote_eye` (and `volcap` can be started via double click). 
Sorry for the inconvenience so far.

kwansik, 2021-01-25:
Thanks to your help, I'm now able to connect to single AzureKinect. I had silly errors in RabbitMq configuring:
1. user name is case sensitive: Volumetric vs volumetric
2. somehow an admin user volumetric was set up without permission to "virtual host".

I have multiple AK connected a workstation but remote_eye_k4a.exe always seems to connect one same device only.
1. any way to connect to multiple devices in my set up?

Assuming I'll be able to connect multiple devices soon, I have further question in calibrations:
1. We are in South Korea(Ewha Womans Univesity) and no IKEA boxes available here. So we are planning to
try similar boxes. What are the limitations on boxes I should use? Smaller dimensions would be ok? or ratio of width,height,depth should be same and so on?

Thanks.

ankarako, 2021-01-25:
Hi again,

As I mentioned in my original reply, connecting multiple devices on a single workstation is not supported. (It will work for one device by it won't work with more than one).  This is because our system follows a distributed design and the `remote_k4a` clients can operate with one device only.
However, If not having NUCs is your problem you can also use spare laptops - or any other PC (`remote_k4a`s do not require high processing capacities and we use NUCs because of their convenient size).

For the lack of IKEA packaging boxes, you just need to check the dimensions of the boxes from the IKEA page (look for JATTENE packaging boxes) and try to find (or make) boxes of the same size.

Hope I helped,
Antonis

kwansik, 2021-01-26:
Thanks for your help and I set up a second device on a laptop successfully connected 2 devices.

I wen ahead and installed followings in python 3.8:
opencv-python
tinyobjloader==2.0.0.rc7
OpenEXR (very painful install. had to download whl and place it under folder and give pip full path to it. and path should not have space or non-english character)
scipy
onnxruntime

Did a calibration captures and ran "process". Used default configurations except increasing inner outer loop count to 25 and 20.
However it runs into an error saying no file found:
\Data\Executables\Calibration\multisensor_calibration\Resources\data\model.onnx

Any idea on what could be an issue here? We used similar boxes (smaller and different dimensions).
Thanks.

vladsterz, 2021-01-26:
Hello @kwansik,
Indeed OpenEXR is a kind of painful installation, that's why we've bundled it under a [single script](https://vcl3d.github.io/VolumetricCapture/docs/calibration/installation/). 
I would suggest you to use the installation procedure above using one of the supported python versions, but in any case it _should_ work, once every library is installed.
The missing model can be found [here](https://drive.google.com/file/d/1_SN4kIfJBbhPisl0kIij_S8Cf2vW_lQK/view?usp=sharing) and it should be placed under the path <path_to_volcap>\Data\Executables\Calibration\multisensor_calibration\Resources\data under the name model.onnx .
After this step it should be fine and working. If that is not the case, I suggest you making the default python of your OS being one of the versions {3.5, 3.6, 3.7} and running the installation script provided. 

On the same topic, can you please give us some feedback once you make it running about the results if the calibration accompanied by the boxes information (sizes, maybe some captures, results). We're just curious.

If calibration seems to produce poor results, please refer to [this](https://vcl3d.github.io/VolumetricCapture/docs/calibration/process/) page to try to place the calibration structure with the correct orientation, relative to the cameras. If you still cannot make it work, try to replicate the boxes dimensions more precisely.

vladsterz, 2021-01-26:
__IMPORTANT__
calibration can be performed when __3 or more__ devices are present!!

kwansik, 2021-01-27:
Thanks for quick replies.
By the way,  we only have boxes which sizes are different from those of IKEA JATTENE(smaller, and different ratios of WxDxH).
Do you think it is worth to try with these boxes or we'd better build boxes with exact same size as IKEA JATTENE ?
Thanks.

ankarako, 2021-01-27:
Unfortunately, our calibration method is data driven, so the boxes must have more or less the exact same size as the JATTENE ones,

closed 2021-03-19 by zuru

terry-r123, 2022-08-17:
> Thanks for your help and I set up a second device on a laptop successfully connected 2 devices.
> 
> I went ahead and installed followings in python 3.8: opencv-python tinyobjloader==2.0.0.rc7 OpenEXR (very painful install. had to download whl and place it under folder and give pip full path to it. and path should not have space or non-english character) scipy onnxruntime
> 
> Did a calibration captures and ran "process". Used default configurations except increasing inner outer loop count to 25 and 20. However it runs into an error saying no file found: \Data\Executables\Calibration\multisensor_calibration\Resources\data\model.onnx
> 
> Any idea on what could be an issue here? We used similar boxes (smaller and different dimensions). Thanks.

Hi, I want to know how to connect two devices to one computer~  
Thanks!