VCL3D/VolumetricCapture #43: Device Repository Tool - Assertion failed

martinpro, 2021-02-10:
After starting volcap in background cmd is this issue
[image]

jmc529, 2021-02-10:
Try following these steps: https://github.com/VCL3D/VolumetricCapture/issues/21#issuecomment-622938983

zuru, 2021-02-11:
Did you first create/add any devices to `device_repository.json` ?

martinpro, 2021-02-12:
Hmmm...same with adding device to repository
[image]

martinpro, 2021-02-12:
I am trying it also on another pc with same result.

martinpro, 2021-02-12:
hmmm...moving program to root c:\volcap helps and now works fine. Thank you for trying to help.

closed 2021-02-12 by martinpro