VCL3D/VolumetricCapture #44: Issue with launching volcap.exe

jmc529, 2021-02-10:
Hi all,
I am having an issue launching volcap.exe I am unsure what I am doing wrong...
Dependent program versions:
- python 3.9.1
- erlang 23.2
- rabbitmq 3.8.11

I have installed rabbitmq and set up a volumetric admin user, added a local azure to my device_repo.json, and have one set up remote. All ports are open inbound and outbound: TCP: 5672, 15672 and UDP: 11234, 320 on the local machine hosting rabbitmq and 321 on the remote machine. 

However, when I open volcap.exe it immediately closes. I opened it in VS19's debugger mode and this is what I was able to see in the command prompt

`Connecting to broker: { amqp://volumetric:capture@192.168.40.10:5672}

GL_VENDOR    : NVIDIA Corporation
GL_RENDERER  : Quadro M5000/PCIe/SSE2
GL_VERSION   : 3.2.0 NVIDIA 436.30
GLSL_VERSION : 1.50 NVIDIA via Cg compiler

DDRenderInterfaceCoreGL initializing ...
> DDRenderInterfaceCoreGL::setupShaderPrograms()
> DDRenderInterfaceCoreGL::setupVertexBuffers()
DDRenderInterfaceCoreGL ready!
`
The exception that shows as "unhanded" in VS19 is 

`Unhandled exception at 0x00007FFCC7383B29 in volcap.exe: Microsoft C++ exception: boost::exception_detail::clone_impl<boost::exception_detail::error_info_injector<boost::system::system_error> > at memory location 0x0000001B430FD550.`

Please get back to me with and recommendations or let me know if more details would be beneficial.

zuru, 2021-02-11:
Seems like a networking/socket issue. Can you try with just volcap (no eye running) to see if the app spawns normally?

jmc529, 2021-02-14:
I had tried that and it did not work. I decided to look through all the firewall rules though and noticed erl was being blocked. (Our university has strict firewall rules). I submitted new firewall rules to the IT department and hopefully that solves it.

closed 2021-02-14 by jmc529