VCL3D/VolumetricCapture #45: Point clouds are not showing in volcap viewer

SergeiSamuilov, 2021-02-26:
When cameras are connected, point cloud streams are not displayed. After calibration the problem remains. I've put depthengine_2_0.dll from Azure Kinect SDK v1.4.1 next to k4a.dll on workstation as well as on NUCs. Also I've installed MICROSOFT VISUAL C++ REDISTRIBUTABLE FOR VISUAL STUDIO 2019 (X64) on all PCs. What am I missing?

vladsterz, 2021-02-26:
Hello @SergeiSamuilov ,
I suppose that 
i) color streams are being shown normally 
ii) you have the latest release

ankarako, 2021-02-26:
Hi @SergeiSamuilov,

Thanks for reaching out!
Could you please be more specific?
- How many kinects are you using?
- What is your setup (Do you use NUCs, Laptops??)
- Are you relying on the `remote_eye_service` or connecting manually to the devices?

Best,
Antonis

SergeiSamuilov, 2021-02-26:
Oh, I understood what I was missing. My workstation laptop autoselected integrated GPU for some reason. When I switched back to main GPU, depth stream started to show normally.

closed 2021-02-26 by ankarako