VCL3D/VolumetricCapture #46: Throwing. Assertion triggered.

ganeshoys, 2021-03-05:
Hi Team, 
    
     Greetings of the day, we are  are using  KINECT  AZURE  and trying to Connect the 3 azure kits using (1 -i7 computer as master and 2  i5 computer as eye ), I have followed everything as instructed  in this link  https://vcl3d.github.io/VolumetricCapture/docs/software/#volcap  till volcap and Copied the DEPTH ENGINE DLL too, but while Executing the Volcap.exe ... the  VolCap.exe(application)  Closes with an error showing " Resources\device_repository.json does not exist. Throwing. Assertion triggered.
Assertion failed: false, file D:\_dev\_Projects\urealsenz\immerzion\include\io\device_repository_loader.h, line 107". Please Guide us further.. 
thank you 
ganesh

vladsterz, 2021-03-05:
Hello @ganeshoys ,
seems that you have not created the "device_repository.json" file.
Please have a look [here](https://vcl3d.github.io/VolumetricCapture/docs/configure/) on how to create one why you need one.

ganeshoys, 2021-03-05:
> eems that you have not created the "device_repository.jso

Hello @vladsterz ,

 thank you for the instant reply, will check this now...

closed 2021-03-19 by zuru