VCL3D/VolumetricCapture #48: dev_repo.exe app crashes

meenaldgupta, 2021-03-25:
while Creating a Device Repository for the device realsense camera.  i get following error.... can anyone help.

C:\Users\subsc\Downloads\release_5.0.0\volcap>dev_repo.exe --add device_name --cam_type 1
2021-03-25 17:03:40.929 DEBUG [224] [main@149] Many devices check
Assertion failed: IsArray(), file D:\_dev\_Projects\urealsenz\immerzion\include\deps\rapidjson\document.h, line 1725

Also i could not see any code on the following path "D:\_dev\_Projects\urealsenz\immerzion\include\deps\rapidjson\document.h" , 

Do i have to download any sourcecode for it.

vladsterz, 2021-03-25:
Hello @meenaldgupta ,
can you please give me some insight on the steps you took to add a device?

meenaldgupta, 2021-03-28:
> Hello @meenaldgupta ,
> can you please give me some insight on the steps you took to add a device?

I got the issue fixed... I had kept an empty JSON file in resource folder .... I replaced it with a JSON file and filled all the required information. Thanks

closed 2021-03-28 by zuru

frenchtouchXR, 2021-11-19:
@meenaldgupta What are the required informations ?

frenchtouchXR, 2021-11-19:
I created an empty file too. And it didn't work. I suppressed it and dev_repo.exe created a correct file.