VCL3D/VolumetricCapture #49: System configuration : Check

meenaldgupta, 2021-03-30:
I have the following System configuration.... will it work
Intel xeon processor
16 gb ram
250 gb ssd
500 gb hdd
NVIDIA Quadro 2000
Windows 10 pro

ankarako, 2021-03-30:
Hi @meenaldgupta ,

Thanks for your interest in our project! I am assuming that those are the specifications of your main workstation PC.
Can you give us more details regarding your setup?
How many sensors do you plan to use for your project?

Antonis

meenaldgupta, 2021-03-30:
yes, the query is regarding workstation PC. 
I have the machine with the above configuration... just want to check will it work.
Right now I am trying the setup for one Realsense D415 Camera.... Once done I will continue the same with 4  Kinect 4 Azure

ankarako, 2021-03-30:
I suppose that you PC is OK. You can check the minimum requirements [here](https://vcl3d.github.io/VolumetricCapture/docs/software/#volcap).

closed 2021-03-30 by ankarako