VCL3D/VolumetricCapture #50: Connecting devices/sensors to Volcap

ganeshoys, 2021-04-12:
Hi team, 

 Greeting of the day,  
  as mention in the "https://vcl3d.github.io/VolumetricCapture/docs/configure/", i have created the device_repository.json, and i do have some issue also, as mention in the  site, while opening the VolCap.exe. the device name will automatically listed in VolCap.exe as per device_repository. but its not showing the List of  device.
[image], 

while checking in intel_nuc_led_utils ( Eye-NUC). am getting this Exception
[image]

i was able to find the log from the "remote_eye_service", the log pasted below.
"WARNINGS
	* The manifest for this application does not have a signature. Signature validation will be ignored.

OPERATION PROGRESS STATUS
	* [12-04-2021 02:05:23 PM] : Activation of C:\Capturer\remote_eye_service\remote_eye_service.application has started.
	* [12-04-2021 02:05:23 PM] : Processing of deployment manifest has successfully completed.
	* [12-04-2021 02:05:23 PM] : Installation of the application has started.

ERROR DETAILS
	Following errors were detected during this operation.
	* [12-04-2021 02:05:23 PM] System.Deployment.Application.InvalidDeploymentException (Zone)
		- Deployment and application do not have matching security zones.
		- Source: System.Deployment
		- Stack trace:
			at System.Deployment.Application.DownloadManager.DownloadApplicationManifest(AssemblyManifest deploymentManifest, String targetDir, Uri deploymentUri, IDownloadNotification notification, DownloadOptions options, Uri& appSourceUri, String& appManifestPath)
			at System.Deployment.Application.ApplicationActivator.DownloadApplication(SubscriptionState subState, ActivationDescription actDesc, Int64 transactionId, TempDirectory& downloadTemp)
			at System.Deployment.Application.ApplicationActivator.InstallApplication(SubscriptionState& subState, ActivationDescription actDesc)
			at System.Deployment.Application.ApplicationActivator.PerformDeploymentActivation(Uri activationUri, Boolean isShortcut, String textualSubId, String deploymentProviderUrlFromExtension, BrowserSettings browserSettings, String& errorPageUrl, Uri& deploymentUri)
			at System.Deployment.Application.ApplicationActivator.PerformDeploymentActivationWithRetry(Uri activationUri, Boolean isShortcut, String textualSubId, String deploymentProviderUrlFromExtension, BrowserSettings browserSettings, String& errorPageUrl)
--- End of stack trace from previous location where exception was thrown ---
			at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
			at System.Deployment.Application.ApplicationActivator.PerformDeploymentActivationWithRetry(Uri activationUri, Boolean isShortcut, String textualSubId, String deploymentProviderUrlFromExtension, BrowserSettings browserSettings, String& errorPageUrl)
			at System.Deployment.Application.ApplicationActivator.ActivateDeploymentWorker(Object state)"

Thank you
Ganesh

ganeshoys, 2021-04-12:
Please Guide me to Connect the Setup 
thank you 
ganesh

ankarako, 2021-04-12:
Hi @ganeshoys and thank you for your interest in our work!

It seems that there is a problem with the `remote_eye_service`. 
1.May I ask if you the model of you NUC-PC? Does it have a led ring on its front plate?

ganeshoys, 2021-04-12:
Thank you @ankarako, 

My NUC's model number is "nuc8i5beh". i think this model doesn't have Ring light 
Have Tried the manual Connection. but not yet connected while in pinging from EYE, Remote_eye_Service -Sent Hello, in master computer i didn't received any response to it

ankarako, 2021-04-12:
Just to make sure,

To connect manually each `eye` to `volcap`:
1. First, open `volcap.exe`
2. Then run the cli command (described [here](https://vcl3d.github.io/VolumetricCapture/docs/eyes/#manual-sensor-connection)) on each NUC

ganeshoys, 2021-04-12:
@ankarako  thank you

 i hope i did this .. let me share you the Help Screen shot on Eye
[image]

i do have  question 
My Master (controller IP is 192.168.1.6) does this doing any issue.

ankarako, 2021-04-12:
You will have to run `remote_eye_k4a.exe --address 192.168.1.6`

ganeshoys, 2021-04-12:
Ya Tried it too...

ankarako, 2021-04-12:
Can you open a command line on the main workstation and type `ipconfig`?

1. How many network adapters are there?
2. When `volcap` starts, does the ip displayed there, match any of the ips displayed with `ipconfig`?

ganeshoys, 2021-04-12:
Thank you @ankarako.
 my computer as only one network port
[image]

  I have tried running eye on the same computer and made remote_eye_k4a.exe --address 192.168.1.6 but I didn't  got any device listed on Volcap.exe

[image]

ankarako, 2021-04-12:
Hmmm, have you opened the ports described [here](https://vcl3d.github.io/VolumetricCapture/docs/software/#rabbitmq-setup) and [here](https://vcl3d.github.io/VolumetricCapture/docs/software/#service--eye)?

ganeshoys, 2021-04-12:
Yes Ankarako
[image]

ankarako, 2021-04-13:
Hi again,

Are the NUCs and the main PC connected on the same LAN via a dedicated switch?

zuru, 2021-04-28:
Closing due to inactivity, feel free to re-open to follow up if necessary.

closed 2021-04-28 by zuru

ganeshoys, 2021-06-19:
@zuru  sorry for the delay reply. .. its been lockdown these days in india. so we are able to try and reply... thank you for closing the questions..
@ankarako sorry for the delay reply, will try the dedicated switch and let u know

ganeshoys, 2021-07-06:
hi  Team. i need some help on calibration, i dont know where am missing..
 the process which we followed
 01. My python version is 3.6 and installed using  python.exe install.py.
 02.  made boxes as mention size on site 
[image]

and followed https://www.youtube.com/watch?v=_0MIxWAE6bw (VCL3D-VolCap: Calibrating 3 devices setting 1), 

The Error 
[image]

zuru, 2021-07-06:
Please see if #53 or #42 help and let us know otherwise