VCL3D/VolumetricCapture #51: Crash when first openning volcap

Tongzhou-Yu, 2021-04-21:
the console shows:

zuru, 2021-04-21:
You need to first create the device repository, see here https://vcl3d.github.io/VolumetricCapture/docs/configure/#creating-a-device-repository.

Tongzhou-Yu, 2021-04-21:
Thanks, but error continues:

C:\Users\Admin>C:\Users\Admin\Downloads\release_5.0.0\volcap\dev_repo.exe --add device_name --cam_type 2
2021-04-21 20:03:46.367 DEBUG [7160] [main@184] List of existing devices:
2021-04-21 20:03:46.367 DEBUG [7160] [main@188] Press any key to exit.

[2021-04-21 20:03:47.790] [error] [t=7160] D:\Projects\libs\Azure-Kinect-Sensor-SDK-master\src\dynlib\dynlib_windows.c (115): dynlib_create(). Failed to load DLL depthengine_2_0 with error code: 126
[2021-04-21 20:03:47.790] [error] [t=7160] D:\Projects\libs\Azure-Kinect-Sensor-SDK-master\src\deloader\deloader.cpp (75): deloader_init_once(). Failed to Load Depth Engine Plugin (depthengine). Depth functionality will not work
[2021-04-21 20:03:47.790] [error] [t=7160] D:\Projects\libs\Azure-Kinect-Sensor-SDK-master\src\deloader\deloader.cpp (76): deloader_init_once(). Make sure the depth engine plugin is in your loaders path

Tongzhou-Yu, 2021-04-21:
There are more error come:

C:\Users\Admin>"C:\Program Files\VCL3D VolumetricCapture release_5.0.0\volcap\dev_repo.exe" --add device_name --cam_type 0
2021-04-21 20:09:17.365 DEBUG [20340] [main@111] Many devices check
2021-04-21 20:09:17.367 DEBUG [20340] [main@119] File .\Resources\device_repository.json doesn't exist, will create one.
2021-04-21 20:09:17.640 DEBUG [20340] [main@135] Device "device_name" added.
2021-04-21 20:09:17.876 DEBUG [20340] [main@184] List of existing devices:
device_name
2021-04-21 20:09:17.881 DEBUG [20340] [main@188] Press any key to exit.

[2021-04-21 20:09:21.020] [error] [t=20340] D:\Projects\libs\Azure-Kinect-Sensor-SDK-master\src\dynlib\dynlib_windows.c (115): dynlib_create(). Failed to load DLL depthengine_2_0 with error code: 126
[2021-04-21 20:09:21.020] [error] [t=20340] D:\Projects\libs\Azure-Kinect-Sensor-SDK-master\src\deloader\deloader.cpp (75): deloader_init_once(). Failed to Load Depth Engine Plugin (depthengine). Depth functionality will not work
[2021-04-21 20:09:21.020] [error] [t=20340] D:\Projects\libs\Azure-Kinect-Sensor-SDK-master\src\deloader\deloader.cpp (76): deloader_init_once(). Make sure the depth engine plugin is in your loaders path

zuru, 2021-04-21:
As reported here: https://vcl3d.github.io/VolumetricCapture/docs/software/#volcap-setup you need to add the K4A depth engine dll manually since we cannot redistribute it.

Tongzhou-Yu, 2021-04-21:
Once it shows the successful message, there is no device_repository.json really been created.

C:\Users\Admin>C:\Users\Admin\Downloads\release_5.0.0\volcap\dev_repo.exe --add device_name --cam_type 0
2021-04-21 20:31:41.400 DEBUG [18648] [main@111] Many devices check
2021-04-21 20:31:41.401 DEBUG [18648] [main@119] File .\Resources\device_repository.json doesn't exist, will create one.
2021-04-21 20:31:41.672 DEBUG [18648] [main@135] Device "device_name" added.
2021-04-21 20:31:41.882 DEBUG [18648] [main@184] List of existing devices:
device_name
2021-04-21 20:31:41.883 DEBUG [18648] [main@188] Press any key to exit.

zuru, 2021-04-21:
You can double-check if it is available by running `volcap.exe`. 
The destination is in `./Resources` as the log indicates.

Tongzhou-Yu, 2021-04-21:
I have checked, the volcap.exe shows the same error:"Resources\device_repository.json does not exist...."
And there is no device_repository.json in /Resources.

vladsterz, 2021-04-22:
Hello @Tongzhou-Yu ,
have you tried running the command prompt in administrator mode?

Tongzhou-Yu, 2021-04-22:
Yes, it works. But the new problem comes, there is nothing shown in the list of the available device on the Devices Widget.

zuru, 2021-04-23:
For the [automatic sensor connection](https://vcl3d.github.io/VolumetricCapture/docs/eyes/#automatic-sensor-connection) you need to set up the [broker](https://vcl3d.github.io/VolumetricCapture/docs/software/#rabbitmq-setup) and the [services](https://vcl3d.github.io/VolumetricCapture/docs/software/#service--eye) at each node. 
Otherwise, you can connect sensors [manually](https://vcl3d.github.io/VolumetricCapture/docs/eyes/#manual-sensor-connection) and they will show up as long as the networking is correctly set up.

zuru, 2021-05-21:
Closing due to inactivity, reopen if needed.

closed 2021-05-21 by zuru