VCL3D/VolumetricCapture #52: Unknown device stream when manually connecting

Ethan-N, 2021-04-30:
Hello, I have successfully gotten the cameras streaming once, but after I rebooted after that run, no matter what I do it seems like I run into the same error from the remote_eye_rs2 where it first has an error saying "Unknown device stream (Color-Depth) when creating a frame source." and then closes the process after saying "No frames received, resetting device hardware." Please let me know what potential fix there is for this. I looked over my setup process and have re-installed the project but because it has worked before I don't think the issue is from how I set it up. Thanks!
[image]

zuru, 2021-04-30:
Hi Ethan,

try the first bullet from the [troubleshooting](https://vcl3d.github.io/VolumetricCapture/docs/eyes/#troubleshooting) section.
You basically need to change a USB controller for the device and it will most likely work.
Try without setting master/slave syncing as well.

zuru, 2021-05-21:
Closing for now due to inactivity, reopen if the issue persists.

closed 2021-05-21 by zuru