VCL3D/VolumetricCapture #53: Could not load file or assembly (remote_eye_service.exe)

rimand, 2021-05-05:
I found problem about remote eyes setup.
I opened install_remote_eye_service.bat and than show this message.

[image]

I guess it will be about enable remote desktop ,i can't enable remote desktop because i use window home edition. 
How can I solve this problem?

ankarako, 2021-05-05:
Hi @rimand, 

Thank you for your interest in our project!

Unfortunately, we have never tested `volcap` on Windows 10 Home machines, so I am not exactly sure how to solve your problem.
However, I don't think that remote desktop has anything to do with it (I am not 110% sure though.)

1. The first thing that comes to mind, is that you didn't run the installation script with admin rights.
2. Another thing, could be that some `ENVIRONMENT VARIABLE`s are not in your machine's `PATH`. You can try to install MS Visual Studio (if you don't have it already installed) and run the installation script though Visual Studio's Command prompt.
3. Finally, you can skip the `remote_eye_service` (and not be able to connect automatically with the `remote_eye`s) an try [manual connection](https://vcl3d.github.io/VolumetricCapture/docs/eyes/#manual-sensor-connection)

rimand, 2021-05-05:
Thankyou,now i see a device connected on rabbitmq server.
But i found a new problem message follow under picture,where I turned off the firwall and opened everything with admin.
Is it because I running volcap and remote_eye_k4a on the same machine?
[image]
how to fix it?

ankarako, 2021-05-05:
Is there a chance that you didn't (open)[https://vcl3d.github.io/VolumetricCapture/docs/software/#rabbitmq-setup] the rabbitmq port required for messaging?

rimand, 2021-05-05:
Thankyou.rabbitmq can work.but i have new problem.
After I opened volcap and remote_eye_k4a. When I click to connect the program remote_eye_k4a It just falls off. And I noticed why it refers to drive D, where I don't have Drive D.

[image]

zuru, 2021-05-05:
Have you downloaded and added the depth engine dll as mentioned [here](https://vcl3d.github.io/VolumetricCapture/docs/software/#service--eye) ?

rimand, 2021-05-07:
Thank it can solve this.
> Have you downloaded and added the depth engine dll as mentioned [here](https://vcl3d.github.io/VolumetricCapture/docs/software/#service--eye) ? 

Now I can connect 4 remote eyes together (without  remote_eye_service) I found a new problem, 
I was unable to calibration, even though it said it was completed, and when I hit load latest, nothing happened. What should I do?

[image]

[image]

`
g2o::SparseOptimizer::optimize: 0 vertices to optimize, maybe forgot to call initializeOptimization()
g2o::SparseOptimizer::initializeOptimization: Attempt to initialize an empty graph
`

ankarako, 2021-05-07:
That great!
Did you install StructureNet as described [here](https://vcl3d.github.io/VolumetricCapture/docs/calibration/installation/#installing-the-calibration-software)?

rimand, 2021-05-07:
Yes, I installed it, but I have a problem installing the module tinyobj.

[image]

So I went to edit the file install.py by comment to install tinyobj and then I installed. tinyobjloader 2.0.0rc7 instead.

[image]

The result is that this is not calibation.

[image]

vladsterz, 2021-05-07:
@rimand ,
sadly I am not able to recreate your issue on my computer. My first cue would be to try to install the deployed tinyobjloader package (you can find it under the directory __volcap\Data\Executables\Calibration\multisensor_calibration\Resources\external\tinyobjloader__ ) by navigating through a command prompt to this directory and executing __pip install .__ (be sure to uninstall the previous one) .
Please provide us with your system's python version.

rimand, 2021-05-07:
I am using python version 3.7.9.
I tried uninstalling tinyobjloader and reinstalling it according to the file location that you can tell the result like this.

[image]

vladsterz, 2021-05-07:
This is quite odd! I would try [this](https://github.com/tbenthompson/cppimport/issues/2#issuecomment-812986319) one, after uninstalling pybind11

rimand, 2021-05-07:
I uninstalled all the modules and reinstalled StructureNet, but I fixed it. requirements.txt A little as follows

[image]

It allows me to python install.py it all at once. 
Right now I have a module like this:

[image]

But the calibation problem is still posting the same message.

[image]

After pressing load latest, nothing happens.

vladsterz, 2021-05-07:
Nice addition.
As the log shows, It has loaded 0 pointclouds for some reason!
Do you press __capture__ prior to __calibrate__ ?
If not can you please give us an insight on the content of the directory  __volcap/Data/Calibrations__ ?

rimand, 2021-05-07:
I press capture before calibrate. The file data from the directory you tell is as follows.

[image]

vladsterz, 2021-05-07:
Could you please share the data with us? I would like to have a closer look and run the calibration procedure on my machine!

rimand, 2021-05-07:
Sure.
https://drive.google.com/drive/folders/1Emyp9q7ANSgq8qdpl6zmZBnk-YRldNt_?usp=sharing

ankarako, 2021-05-07:
Seeing the boxes that you utilize, I have note that:
1. The boxes in your images do not have the correct placement. Please see [this](https://vcl3d.github.io/VolumetricCapture/docs/calibration/process/#calibrating-a-volumetric-setup).
2. In addition, It seems that you do not use the boxes we recommend. You should be aware that if you use another kind of packaging boxes they need to match the exact dimensions of [these ones](https://vcl3d.github.io/VolumetricCapture/docs/calibration/assembly/#assembling-the-calibration-structure).

vladsterz, 2021-05-07:
Seems that we support device names without underscores, fact that we should add to the wiki I guess.
You can use the __device_repository__ tool which supports a rename command, and rename your devices to names that have no underscores! (You could also rename them directly in the device_repository.json file)
Then re-capture and calibrate!
Here is the sample log with the data you send me, after I renamed the devices to match devices to our device_repository.
[image]

rimand, 2021-05-07:
I rename every new kinect name. It's KA01, KA02, .. but the message appears the same.

[image]

rimand, 2021-05-07:
Today Thankyou vary much
I'll try to change my Vocap machine in case it will produce different results.

rimand, 2021-05-08:
After I installed the program on the other computer In this case, I am using python 3.8.10 so I can calibrate it., but I have to check if there are any other factors because the new device I installed is very messy. What I do further is that I only add python to it.

[image]

rimand, 2021-05-08:
I want to know if your software can do the image clean like this in realtime or do I have to go to the Post process?
(Clen here means the picture is not missing, such as a loose head hair.)

[image]

closed 2021-05-21 by rimand

zuru, 2021-05-21:
Currently volcap only captures the raw data without any post-processing.