VCL3D/VolumetricCapture #59: Source code available?

machineroom, 2021-09-20:
Hi, this looks like a very useful project. Do you plan to make the source code available so the community can contribute? e.g. I'd be interested in a Linux port.
Also, is this still being maintained?

Thanks!

vladsterz, 2021-09-20:
Hello @machineroom and thank you :) .
At this time this project is discontinued so no updates will be made in the near future.

machineroom, 2021-09-27:
Thanks for the reply and understood. Would you be prepared to open the source code though?

vladsterz, 2021-09-28:
At this moment we have no such plans.

zuru, 2021-09-29:
Unfortunately, the project is undergoing a low/no activity period, so at the moment there are no such plans. Should it get back on track this is something we will surely consider.

closed 2021-09-29 by zuru