VCL3D/VolumetricCapture #61: Volcap.exe crashing

frenchtouchXR, 2021-11-19:
I get this error from the cmd. 

Resources\device_repository.json does not exist. Throwing. Assertion triggered.
Assertion failed: false, file D:\_dev\_Projects\urealsenz\immerzion\include\io\device_repository_loader.h, line 107

I7 - 32go RAM - NVIDIA GTX 1080 - Windows family

I have Python installed and I installed RabbitMQ

frenchtouchXR, 2021-11-19:
OK I think I need to create the device repository and set up RabbitMQ as written here. 
https://vcl3d.github.io/VolumetricCapture/docs/configure/

frenchtouchXR, 2021-11-19:
I set up RabbitMQ then created an empty device_repository.json with this line : 

I tried to add a device with this command line : .\dev_repo.exe --add azure0 --cam_type 0 
Dev_repo crashes and I get this error message : 

2021-11-19 16:32:58.941 DEBUG [13092] [main@111] Many devices check
Assertion failed: IsArray(), file D:\_dev\_Projects\urealsenz\immerzion\include\deps\rapidjson\document.h, line 1725

zuru, 2021-11-19:
Please check if it is the same issue as with #48

closed 2021-11-20 by frenchtouchXR