VCL3D/VolumetricCapture #62: No camera detected in the widget

frenchtouchXR, 2021-11-19:
I added 2 Kinect Azure cameras in the device repository, I start the volcap.exe but the devices are not detected. Could you please help ?

zokin, 2021-11-19:
Please check if the [services](https://vcl3d.github.io/VolumetricCapture/docs/software/#service--eye) have been [configured](https://vcl3d.github.io/VolumetricCapture/docs/eyes/#automatic-sensor-connection) correctly and are up and running.

frenchtouchXR, 2021-11-19:
Is enabling remote desktop necessary ? As I work on a family version of windows this option is not available.

zokin, 2021-11-19:
Not really, it is just for accessing headless mini-PCs easily.

frenchtouchXR, 2021-11-20:
I had the devices detected but none is streaming (tried with 2 azure kinect)
[image]
.

frenchtouchXR, 2021-11-20:
I managed to make it work by copying depthengine_2_0.dll to the Capturer directory.

frenchtouchXR, 2021-11-20:
Now trying to get 2 Kinect Azure streaming on the same computer.

ankarako, 2021-11-20:
Is the main volcap application running on the same machine that the kinects are connected (and therefore, the `remote_eye_k4a.exe`s run on)?

frenchtouchXR, 2021-11-20:
The app runs on the same machine than the kinects are connected on (when I start volcap.exe remote_eye_k4a.exe stops). I notice the 2 kinects have the same IP. 192.168.1.11 maybe it has something to do.

frenchtouchXR, 2021-11-20:
Remote eye send me this message : 

> Logging to: C:\Capturer\bin
> Connected K4A device /w serial no: 000501194612
> Connecting to RabbitMQ broker: amqp://volumetric:capture@195.251.117.104:5672 # Handshake
> Sent hello from 000501194612
> Waiting for welcome (press 'q' to quit) ... 

If I run a second instance the new one crashes.

frenchtouchXR, 2021-11-20:
Does only one kinect by machine is allowed ?

ankarako, 2021-11-20:
If everything is connected on the same PC, it makes sense for the kinects to show up with the same IP. However, you will have a problem with this setup, due to the `remote_eye_service`. When a `remote_eye.exe` is running, and volcap opens, the service will kill the remote_eye application to spawn a new one. So that's your issue.
Currently, volcap cannot support `N` kinects connected on the same device.

frenchtouchXR, 2021-11-20:
OK I'm going to try to connect another kinect to my 2cd machine.

frenchtouchXR, 2021-11-20:
I got my 2cd machine working with remote eye. 
Machine 0 with Kinect 0 and Volcap
Machine 1 with Kinect 1 
I opened in and out the 11245 port for both machine. 
The machine 0 controls and receive the Kinect 1 streaming. The problem is now the Kinect 0 doesn't stream anymore. Is it possible to have Kinects on both the master machine and the slave machine ?

frenchtouchXR, 2021-11-20:
I managed to do so by pluging in the Kinect 0 first, make it recognized and then pluging in the kinect 1 and refresh then connect all.

zuru, 2021-11-22:
Sometimes either a refresh is required, or switching USB ports, and if all fails perhaps a volcap restart may fix this :)

closed 2021-11-22 by zuru