VCL3D/VolumetricCapture #65: Volcap.exe crashing

GuilhermeMEESantos, 2021-12-03:
Good morning. 

I have been having a problem with volcap.exe always crashing. 

It gives me this: Assertion failed: IsArray(), file D:\_dev\_Projects\urealsenz\imerzion\include\deps\rapidjson\document.h, line 1725

I know that the error is in the need for a device_repository.json file in the Resources folder. 

My question is: How do I create it and what do I need to write in it?

zokin, 2021-12-03:
Please check this part of the docs: [creating a device repository](https://vcl3d.github.io/VolumetricCapture/docs/configure/#creating-a-device-repository)

GuilhermeMEESantos, 2021-12-03:
I have read that paragraph already but I still don't know how to do it. Could you provide an example?
Sorry for bothering you like this but this is all new to me. I never used JSON files.

GuilhermeMEESantos, 2021-12-03:
And thank you for the instant reply.

GuilhermeMEESantos, 2021-12-03:
[image]

closed 2021-12-03 by GuilhermeMEESantos