VCL3D/VolumetricCapture #66: Volcap.exe

GuilhermeMEESantos, 2021-12-03:
While starting volcap.exe it gave me this:
Assertion failed: IsArray(), file D:_dev_Projects\urealsenz\imerzion\include\deps\rapidjson\document.h, line 1725

My question is: How do I use the device repository tool to create the device repository file?

Could someone provide examples or even a step by step guide?

zokin, 2021-12-03:
Ok, for each device:
- Plug it
- Run the `add` command using a distinct name
- The `.json` gets updated
- Unplug it so that the next one can be plugged

Let us know where this process breaks and how so that we can be of more help.

GuilhermeMEESantos, 2021-12-03:
Ok, but where do I run the add command? On the volcap.exe?

GuilhermeMEESantos, 2021-12-03:
Or on the dev_repo.exe?

GuilhermeMEESantos, 2021-12-03:
The thing is that both crash. Volcap.exe with the command I sent above and the dev_repo.exe doesn't even start

zokin, 2021-12-03:
Even if you run `dev_repo.exe -h`? 

[image]

GuilhermeMEESantos, 2021-12-03:
How do I run it?

GuilhermeMEESantos, 2021-12-03:
On a command prompt? I tried and it doesn't recognize the command

zokin, 2021-12-03:
On which path do you run it from?

GuilhermeMEESantos, 2021-12-03:
On the C:\Windows\system32

GuilhermeMEESantos, 2021-12-03:
How do I change the path?

GuilhermeMEESantos, 2021-12-03:
Ok, already have C:\volcap

zokin, 2021-12-03:
Please resolve these issues and check the documentation again as well as past issues to move forward with settings this up.

GuilhermeMEESantos, 2021-12-03:
I tried adding a camera and it shows me this:

C:\volcap>dev_repo.exe -- add device_name --cam_type 1
The following arguments were not expected: -- add device_name --cam_type 1
Run with --help for more information.
[2021-12-03 12:10:49.319] [error] [t=12664] D:\Projects\libs\Azure-Kinect-Sensor-SDK-master\src\dynlib\dynlib_windows.c (115): dynlib_create(). Failed to load DLL depthengine_2_0 with error code: 126
[2021-12-03 12:10:49.320] [error] [t=12664] D:\Projects\libs\Azure-Kinect-Sensor-SDK-master\src\deloader\deloader.cpp (75): deloader_init_once(). Failed to Load Depth Engine Plugin (depthengine). Depth functionality will not work
[2021-12-03 12:10:49.321] [error] [t=12664] D:\Projects\libs\Azure-Kinect-Sensor-SDK-master\src\deloader\deloader.cpp (76): deloader_init_once(). Make sure the depth engine plugin is in your loaders path

zokin, 2021-12-03:
There is a space between the dashes and add.

GuilhermeMEESantos, 2021-12-03:
Now it gives the same error:

Assertion failed: IsArray(), file D:\_dev\_Projects\urealsenz\immerzion\include\deps\rapidjson\document.h, line 1725

GuilhermeMEESantos, 2021-12-03:
I have the empty file created though

GuilhermeMEESantos, 2021-12-03:
I deleted it and it created the file. 
Thank you for your attention and for your help!
Thank you so much!

Hope you have a great day

closed 2021-12-03 by GuilhermeMEESantos