VCL3D/VolumetricCapture #67: remote_eye_service.exe doesn't install

GuilhermeMEESantos, 2021-12-06:
Good afternoon. 

I am trying to connect an Intel Realsense D415 to an Intel NUC8i7BEH and in turn connecting directly to my personal pc with an ethernet cable.

After configuring the needed ports, TCP 5672 and 15672 and UCDP 11234, and creating the Capturer file and putting the bin and remote files in it, I tried installing the remote eye service with administrative rights but it gave this:

**Exception occurred while initializing the installation:
System.IO.FileLoadException: Could not load file or assembly 'file:///C:\Capturer\remote_eye_service\remote_eye_service.exe' or one of its dependencies. Operation is not supported. (Exception from HRESULT: 0x80131515).

C:\Capturer\remote_eye_service>net start remote_eye_service
The service name is invalid.**

Could it be the fact that I am using Windows 10 Education on the Intel NUC? Or is it something else missing? Could it be the fact Python is not installed in the NUC?

Or alternatively, should I try and connect the devices manually? In that case, I would need to go to the device with the Volcap and use the remote_eye_{k4a|rs2}.exe command. With what directory can I use the remote_eye_{k4a|rs2}.exe command on the command prompt?

ankarako, 2021-12-06:
Hi @GuilhermeMEESantos and thank you for your interest in our project.

First, Windows Education edition could be the reason that the service cannot be installed, however, I'm not 100% sure because we haven't tried it.
To connect manually to the remote eyes, you should first open volcap on your main PC, and then run the remote_eye_rs2.exe (since you are using Realsense sensors) on the NUC through the command line. Please check [here](https://vcl3d.github.io/VolumetricCapture/docs/eyes/#manual-sensor-connection) for info on how to do this.

GuilhermeMEESantos, 2021-12-06:
But I can only start Volcap after adding a device to the device repository JSON file, right?

So I first do that and then run the remote_eye_rs2.exe on the NUC?

GuilhermeMEESantos, 2021-12-06:
It doesn't let me add a new device because:

FATAL [12596] [main@145] "-add" command failed, no IntelRealsense devices connected, please connect one.

But both my computer and the Intel NUC are comunicating with each other, at least I can ping between.

GuilhermeMEESantos, 2021-12-06:
On the NUC the command prompt sent the HELLO but it is taking a long time to wait for the response and in the volcap command prompt it shows this in yellow:

Connecting to broker: { amqp://volumetric:capture@192.168.0.101:5672}

ankarako, 2021-12-06:
Did the device appear on volcap when you ran remote_eye_rs2.exe?

GuilhermeMEESantos, 2021-12-06:
It did let me add it to the device_repository.

Actually it even created the JSON file and added it.

[image]

GuilhermeMEESantos, 2021-12-06:
But it doesn't show on the program
[image]

GuilhermeMEESantos, 2021-12-06:
Most likely because it is still trying to make the connection

ankarako, 2021-12-06:
The device_repository, does not have anything to do with either seeing the device on volcap, or connecting with it. Please check first volcap's [workflow](https://vcl3d.github.io/VolumetricCapture/docs/volcap/) to understand how you should operate the program.

GuilhermeMEESantos, 2021-12-06:
I think I may know the problem. Volcap is not connecting to the RabbitMQ broker. I will try and open Volcap with this command:

volcap.exe -b <rabbit_mq_broker_local_ip> -u <rabbit_mq_username> -p <rabbit_mq_password>

As it is shown [here](https://vcl3d.github.io/VolumetricCapture/docs/volcap/#preparation).

GuilhermeMEESantos, 2021-12-06:
But how do I see what ip is rabbit configured with?

GuilhermeMEESantos, 2021-12-06:
Or is it associated with my pc's ip?

ankarako, 2021-12-06:
Where did you install RabbitMQ?

GuilhermeMEESantos, 2021-12-06:
On both the NUC and my pc

GuilhermeMEESantos, 2021-12-06:
Was is to install RabbitMQ only in the pc where Volcap will be?

ankarako, 2021-12-06:
Well, you should install it only on your main PC. Please check the instruction [here](https://vcl3d.github.io/VolumetricCapture/docs/software/).

GuilhermeMEESantos, 2021-12-06:
Oh, as I was reading I thought it was to install both in the Volcap pc and on the eyes.
So I will uninstall rabbit on the NUC

GuilhermeMEESantos, 2021-12-06:
What about erlang? Should I uninstall it as well on the NUC?

ankarako, 2021-12-06:
Yeah, erlang is needed only for rabbit mq

GuilhermeMEESantos, 2021-12-06:
It's still attempting the connection. But I noticed that the volcap app tries to connect to the RabbitMQ with my pc's ip but the remote_eyes_rs2 attempts with an ip address that is neither my pc's ip neither NUC's

GuilhermeMEESantos, 2021-12-06:
Do I need to configure RabbitMQ management again?

ankarako, 2021-12-06:
Do you have more than one network adapters on your PC (wireless, vEthernet)?

GuilhermeMEESantos, 2021-12-06:
[image]

GuilhermeMEESantos, 2021-12-06:
I just have these 4 and 2 are disconected
I also had a virtual ethernet adapter but that is disabled so it must not be it.

GuilhermeMEESantos, 2021-12-06:
No, it is a different one

GuilhermeMEESantos, 2021-12-06:
Oh so I need to use the remote_eye command on the NUC and set the ip, like it shows [here]([image]), right?

GuilhermeMEESantos, 2021-12-06:
[image]

GuilhermeMEESantos, 2021-12-06:
I think it worked

closed 2021-12-06 by ankarako

GuilhermeMEESantos, 2021-12-06:
Thank you for your help and for taking the time to help me. Ever since I started trying to create a mocap system using your open-source code that I noticed that the creators were very helpful and took the time to help. 

Wish you a continuation of a good day and of good work.

Best regards,

GuilhermeMEESantos