VCL3D/VolumetricCapture #68: Configuration: adding REALSENSE DEPTH CAMERA D415 to device repository

mtakhsha94, 2021-12-06:
Hi Everyone,
I have 4 Realsense depth camera D415. only one of the camera is added to the device repository successfully. the other ones don't add to the device repository and I also don't get any error regarding this issue. does anybody have the same problem?
I appreciate it if you help me in this regard and give me a solution to this issue.

ankarako, 2021-12-06:
Could you check the `device_repository.json` to see that the the devices are not added?

mtakhsha94, 2021-12-06:
I checked that and it is just the first camera. I also deleted the device_repository.json and created new one and tried again. but they are not added to the device repository.

ankarako, 2021-12-06:
Could you share the commands you used for adding the devices?

mtakhsha94, 2021-12-06:
Every time, I connect just 1 camera and use this codes seperately for each of them:
for camera 1:
dev_repo.exe --add rs1 -cam_type 1
for camera 2:
dev_repo.exe --add rs2 -cam_type 1
for camera 3:
dev_repo.exe --add rs3 -cam_type 1
for camera 4:
dev_repo.exe --add rs4 -cam_type 1

but the command is just response when I use the first camera. it doesn't work for the other cameras.

ankarako, 2021-12-07:
Hi again,
I cannot seem to find what is wrong for now, however, a workaround could be to create different device_repository files for each device, and then merge them manually. You could try it until we have time to investigate on this.

zokin, 2021-12-08:
While it was not reproducible right from the start, the same issue did manifest after some tries indeed. These devices require meticulous handling when plugging and unplugging them quickly. The way to resolve it is to make sure that you wait until the device manager correctly refreshes (twice, you should also be getting taskbar notifications) and prepares the device before accessing it, otherwise, the driver/controller is left in a corrupted state.

So after this same problem manifested (was getting no entries or file creation), you should unplug the device then replug it and wait till it is functional (check with RS2 viewer and/or the device manager). Sometimes this can be addressed by switching a USB port/controller. Then run the device add command. Then repeat for each device you want to add.

Below is a terminal screenshot of registering two D415s and the accompanying `device_repository.json` file (zipped).

[image]

[device_repository.zip](https://github.com/VCL3D/VolumetricCapture/files/7674961/device_repository.zip)

closed 2021-12-08 by zokin

mtakhsha94, 2022-01-12:
Sorry for the late reply.
I solved the problem by installing the sensor drives and updating the sensors. after that I easily add them to device_repository.jason
Thank you for your help :)