VCL3D/VolumetricCapture #71: remote_eye_service won't start to run in the background

ttkrpink, 2021-12-22:
Hello, I followed the following steps:

1. inbound UDP 11234,320,321
2. outbound UDP 11234,320,321
3. Run the install_remote_eye_service batch file as admin to install
4. remote_eye_service started to run in the background, with a PID. Then it stopped.
5. Set the remote_eye_service.exe as "Run always as admin" in the compatibility mode.
6. reboot the computer, remote_eye_service is not running in the background. 
7. Start the service manually, it grabs a PID, then stops again.
8. reinstall vc_redist.x64, try the 3-7 again.

I didn't find the remote_eye_service log file. So, what could be the problem? What am I missing here? I did try uninstall_remote_eye_service  as admin before installing.

Thanks for your help.

ankarako, 2021-12-23:
Hello,

Regarding the service's log file, did you check the "Capturer" folders on the PC that the remote_eyes run on?

ttkrpink, 2021-12-23:
Thanks for your reply. I reboot all my computers, the eye controller, and the broker. Now, the problem goes away.

Now there is a new problem, I am not sure I should open a new issue or stay on this one. Anyway, here it goes:

I have three eye controllers. Two of them can be identified automatically by the volcap host. For the last one, I have to manually use "remote_k4a.exe --address 123" to access. Where do I start to troubleshoot? 

Thanks for your help

ankarako, 2021-12-23:
There should be something wrong with the services that runs on the eye controller. First, you can check if the services has started.

ttkrpink, 2021-12-24:
Thanks for your help. All problems are solved by reinstalling and rebooting.

closed 2021-12-24 by ttkrpink