VCL3D/VolumetricCapture #77: I connect as required, but I can't see the image

yuwuzhi1989, 2022-06-08:
Glad to see such a project, it's awesome, I tried to buy the equipment and install it. But there are some problems.
I connected a Kinect dk on the local computer, and configured python and rabbitmq, then opened the volcup software, I can see the Kinect I configured in the Devices column, I selected connect, and I can see it in connected, but There is no transfer, no image is visible in the small pop-up window.
[image]
Please help me, I've tried all day and can't solve this problem. thanks

ankarako, 2022-06-08:
Hi and thank you for your interest in our work.
Did you configure your firewall settings?
Is the remote_k4a running or crashed?
If I understood correctly you have the kinect connected on the same PC, on which volcap is running?

yuwuzhi1989, 2022-06-08:
Think you。
I think remote_k4a running is ok.
remote_eye.log :
2022-05-31 20:39:09.090 INFO  [1632] [main@68] Logging to: C:\Capturer\bin\remote_eye.log
2022-05-31 20:39:09.952 INFO  [1632] [main@75] Connected K4A device /w serial no: 000768714212
2022-05-31 20:39:09.952 INFO  [1632] [main@80] Connecting to RabbitMQ broker: amqp://volumetric:capture@192.168.5.11:5672 # Handshake
2022-05-31 20:39:10.986 INFO  [1632] [remote::handshake::say_hello@50] Sent hello from 000768714212
2022-05-31 20:39:10.986 INFO  [1632] [main@239] Waiting for welcome (press 'q' to quit) ...
2022-05-31 20:40:02.942 INFO  [8252] [main::<lambda_1434f7e39b4c52ca1680e8fd13eafafa>::operator ()@118] Creating networking components...
2022-05-31 20:40:02.944 INFO  [8252] [main::<lambda_1434f7e39b4c52ca1680e8fd13eafafa>::operator ()@126] Streaming channel opened.
2022-05-31 20:40:02.974 WARN  [8252] [local::led_controller::led_controller@47] NUC Led controller app is available @ C:\Capturer\bin\intel_nuc_led_utils.exe.
2022-05-31 20:40:02.974 WARN  [8252] [local::led_controller::led_controller@55] Spawning powershell.exe -Command Start-Process 'C:\Capturer\bin\intel_nuc_led_utils.exe' 'ALWAYS_ON GREEN' -Verb runAs
2022-05-31 20:40:03.339 INFO  [8252] [main::<lambda_1434f7e39b4c52ca1680e8fd13eafafa>::operator ()@136] Command channel opened.
2022-05-31 20:40:03.339 INFO  [8252] [main::<lambda_1434f7e39b4c52ca1680e8fd13eafafa>::operator ()@157] Initializing syncer on C:\Capturer\bin\PTPSync_slave.exe ...
2022-05-31 20:40:03.341 INFO  [8252] [main::<lambda_1434f7e39b4c52ca1680e8fd13eafafa>::operator ()@163] Syncer initialized.
2022-05-31 20:40:04.217 INFO  [8252] [main::<lambda_1434f7e39b4c52ca1680e8fd13eafafa>::operator ()@182] Initializing device w Color-Depth streams.
2022-05-31 20:40:04.217 INFO  [8252] [main::<lambda_1434f7e39b4c52ca1680e8fd13eafafa>::operator ()@189] Loading preset <accuracy-100um> from C:\Capturer\bin\Resources\accuracy-100um.json
2022-05-31 20:40:04.268 INFO  [8252] [main::<lambda_1434f7e39b4c52ca1680e8fd13eafafa>::operator ()@194] Acquiring device parameters...
2022-05-31 20:40:04.318 INFO  [8252] [main::<lambda_1434f7e39b4c52ca1680e8fd13eafafa>::operator ()@203] Device parameters initialized from configuration file: C:\Capturer\bin\Resources\accuracy-100um.json

Firewall is turned off
[image]

Yes，I have the kinect connected on the same PC, on which volcap is running。

yuwuzhi1989, 2022-06-08:
I originally used two computers。one of which was connected to another Kinect。volcup was installed on this machine, and a Kinect was installed. I hope to complete the effect of the face through two Kinects.

yuwuzhi1989, 2022-06-08:
Sorry. I misunderstood before.I opened port: 5672,15672,11234 on Windows Firewall ,I did it according to the instructions.

ankarako, 2022-06-08:
Can you open the rabbitmq dashboard to check if the remote_eye* application sends any data over the network?

yuwuzhi1989, 2022-06-08:
First,
[image]

Second,I open the rabbitmq dashboard to check, but I don't see data transfer。
[image]
[image]

When I click Connect on volcap
[image]

Third, remote_eye is opened.
[image]

ankarako, 2022-06-08:
Hmm, the last screenshot is from the remote_eye_service, which is supposed to open the remote_eye application.
Since you are connecting the kinect on the same PC running volcap, there could be an issue with the service. Could you please stop the service, then open volcap, and then try to run the remote_eye application manually (as described [here](https://vcl3d.github.io/VolumetricCapture/docs/eyes/#manual-sensor-connection))

yuwuzhi1989, 2022-06-08:
First I turned off remote_eye, then I followed the sequence of the picture below.
Is this right?
[image]
[image]
[image]
I still can't see the picture

yuwuzhi1989, 2022-06-08:
I have done the same configuration on another computer, so that volcup and remote_eye are not on the same computer, they are connected by network cable, I now use the command volcap.exe -b 192.168.5.21 -u volumetric -p capture to execute , still can't see the image
[image]
[image]

Thank you so much for your help all the time

ankarako, 2022-06-08:
Hmm, I don't think that you followed the correct steps.
You should do the following steps:
1. first Stop the remote_eye_service (i.e. Open "Task Manager", go to "Services" tab, find "remote_eye_service", right-click and select "Stop")
2. Open volcap.exe by double-clicking.
3. Open remote_eye_k4a.exe by using the command line with the instructions provided in the link in my previous comment.

yuwuzhi1989, 2022-06-10:
Now,I remove the remote_eye_service 
then ,Open volcap.exe by double-clicking.

yuwuzhi1989, 2022-06-10:
Thanks very much ,I successed.
Because ‘Enable High Performance Settings ’ Setup failed before

yuwuzhi1989, 2022-06-10:
How can I connect 2 Kinects on one computer？
I don't get the method

closed 2022-07-04 by zokin