VCL3D/VolumetricCapture #78: Onnxruntime Error: Protobuf parsing failed

dbrice01, 2022-06-14:
When selecting "Process" in the calibration stage in a setup with 4x Kinect cameras and default settings I receive the following error: 

D:\32\s\onnxruntime\core\session\inference_session.cc:220 onnxrunime::Inference_Session::Inference_Session statis.IsOK() was false. Given model could not be parsed while creating inference session. Error message: Protobuf parsing failed

Any ideas?

Thanks.

dbrice01, 2022-06-14:
[image]
[image]

zuru, 2022-06-14:
It may be related to `protobuf` breaking backwards compatbility. Try downgrading with `pip install protobuf==3.20.1`

dbrice01, 2022-06-15:
Thank you for your quick reply and suggestion.

[image]

It seems that the 3.20.1 distribution is missing from pip. Would you recommend I try to install 3.20.1 with an alternative method, or should I try and use one of the available distributions? 

Also, pip version is 21.3.1

Thanks

dbrice01, 2022-06-15:
Since my last post: I've verified that protobuf version 3.20.1 is installed (after switching from python 3.6 -> 3.7), however the error, "protobuf parsing failed" remains.

Thanks

dbrice01, 2022-07-04:
Has anyone got any ideas on troubleshooting this?  All of the required packages seem to be in place during the StructureNet install. Thanks

zokin, 2022-07-04:
Not sure whats up but can you try download the model from gdrive (see installation.txt) and then running these:

[image]

On my side it seems to be ok with;

```
onnx                         1.8.1
onnxruntime                  1.6.0
protobuf                     3.19.1
```

dbrice01, 2022-08-25:
Issue was resolved through resetting the Host PC environment with the package versions provided in the comments above. 
Thank you.

closed 2022-08-25 by dbrice01