VCL3D/VolumetricCapture #79: dev_repo.exe not creating device_repository.json

dhorspool, 2022-06-16:
Each time I try to add a new device by using dev_repo.exe to create a device repository the Json is not created in the resources. The intel realsense cameras are working as I have their viewer working as I try to add them. I am trying only one camera at a time. 

I have tried moving \volcap to C:\. Do I need to create a device_repository.json beforehand? this is not clear from documentation. 
[image]

alexd314, 2022-06-16:
Hello,
What is the model of you Intel RealSense sensors ?

dhorspool, 2022-06-16:
When running command prompt in Admin mode, dev_repo progresses further bu still does not appear to create a device_repository.json file 
[image]

ankarako, 2022-06-16:
Could you please check if `device_repository.json` is created somewhere else (such as volcap's binary dir - i.e. the parent directory of the "Resources" folder) this was an issue, and I can't remember if it is resolved.

dhorspool, 2022-06-16:
Does not appear to be in the volcap binary dir. 
[image]
I have also tried manually updating the path using "C:\>volcap\dev_repo.exe --path C:\volcap\Resources" .  Also used searched function on file explorer and nothing came up

closed 2022-06-24 by dhorspool