VCL3D/VolumetricCapture #81: Need help for Calibration

ipoirot, 2022-07-06:
Hi,
Glad to see such a project, it's awesome.

I used  6 Kinect and successfully connected them all. When I go to Calibration -> Process, I got no results but it said that the calibration is completed.
[image]

What I have done:
1. I used the boxes with the specific size: 56x33x41 cm.
2. I installed StructureNet as described.
3. I ran these described by zobin:
[image]
it seems to be ok. It means that the installation is correct.  Do I understand correctly?

Could you shed some light into what I am doing wrong? Thanks.

ipoirot, 2022-07-06:
Please check out the attachment for more details.

[device_repository.zip](https://github.com/VCL3D/VolumetricCapture/files/9050549/device_repository.zip)
[22-07-05-12-42-29.zip](https://github.com/VCL3D/VolumetricCapture/files/9050550/22-07-05-12-42-29.zip)
[22-07-05-12-32-52.zip](https://github.com/VCL3D/VolumetricCapture/files/9050551/22-07-05-12-32-52.zip)

zuru, 2022-07-06:
These folders contain 5 image pairs not 6, can you confirm that the 6th sensor is streaming data?

ipoirot, 2022-07-06:
> These folders contain 5 image pairs not 6, can you confirm that the 6th sensor is streaming data?

Thanks for your quick reply.

This file contains 6 image pairs, but also failed
[22-07-03-12-10-09.zip](https://github.com/VCL3D/VolumetricCapture/files/9053135/22-07-03-12-10-09.zip)
.

zuru, 2022-07-06:
The segmentation fails in these examples, please verify that the [instructions](https://vcl3d.github.io/VolumetricCapture/docs/calibration) have been followed:
1. The structure has been assembled correctly
2. The distance of the structure is appropriate
3. The sensors are vertically positioned (does not seem to be the case).

ipoirot, 2022-07-07:
> The segmentation fails in these examples, please verify that the [instructions](https://vcl3d.github.io/VolumetricCapture/docs/calibration) have been followed:
>
> 1. The structure has been assembled [...]

Thanks a lot for your detailed reply, it's very helpful.

ipoirot, 2022-07-12:
I have been followed the instructions for 4 Azure Kinect depth cameras :
1. The structure has been assembled correctly.
2. The distance from the ground of all sensors is about 1.2m
3. All sensors are positioned at a distance from the structure about 2.0m. 
4. All sensors are placed vertically.

Still got this error:
[image]

ipoirot, 2022-07-12:
Please check out the attachment for more details
[device_repository.zip](https://github.com/VCL3D/VolumetricCapture/files/9090172/device_repository.zip)
[22-07-09-02-49-26.zip](https://github.com/VCL3D/VolumetricCapture/files/9090178/22-07-09-02-49-26.zip)

.

zuru, 2022-07-12:
It doesn't seem that the sensors have been positioned vertically as the attached color images from the calibration data folder you provided are placed horizontally. See this docs [page](https://vcl3d.github.io/VolumetricCapture/docs/calibration/process/) for an example on how it would look inside the app, 

Please turn the Kinect4Azure sensors 90 degrees vertically in a counter clock-wise manner when looking at the sensor (see [here](https://vcl3d.github.io/VolumetricCapture/docs/acquisition/)) and try again.

[image]
[image]
[image]
[image]