VCL3D/VolumetricCapture #83: Reduction of error during calibration

dbrice01, 2022-08-25:
I've attempted running the calibration in different environments, with 4 cameras in different positions and so far I have not been able to reduce the size of the errors, resulting in poor camera alignment and transforms. The cameras have been setup following the instructions at: https://vcl3d.github.io/VolumetricCapture/docs/calibration/process/ 

Due to there being no availability of the IKEA JÄTTENE boxes we have laser cut and assembled MDF boxes of identical proportions, could this be the issue? Otherwise are there any suggestions for achieving a more successful calibration?

[image]
[image]

Many Thanks,
Daniel

zuru, 2022-08-25:
Can you please try to rotate the boxes in the opposite direction when assembling the structure and see if that works?
I think that compared to the views depicted in the videos in this [page](https://vcl3d.github.io/VolumetricCapture/docs/acquisition/) the assembly process is not consistent with the instructions [here](https://vcl3d.github.io/VolumetricCapture/docs/calibration/assembly/).

dbrice01, 2022-09-06:
That's exactly what my problem was, thank you for your observation. The calibration errors are now at a suitable level.

closed 2022-09-06 by dbrice01

zuru, 2022-09-06:
One more thing, when packaging the IKEA boxes, the resulting dimensions are typically not the ones reported by the IKEA website. Since you are manufacturing a replica, you should take that into account, and measure the dimensions of the `/Data/Executables/Calibration/multisensor_calibration/Resources/data/subsampled_asymmetric_structure_corrected.ply` which we updated to be in line with the assembled dimensions, and create your replica according to them. This will additionally improve accuracy.

dbrice01, 2022-09-14:
Thank you for your advice. I have not been able to find a precise way to measure the dimensions of the structure from the .PLY file, but will continue to look into it.

ankarako, 2022-09-14:
Hi, you can use meshlab's measuring tool