VCL3D/VolumetricCapture #84: How to get extrinsic

terry-r123, 2022-08-30:
Hi, @zokin @vladsterz 
Thank you for your exciting work!

I would like to know when I only have run `calibration.py` with my own` .pgm` and `.png` files of multiple cameras and got the `calibration.json`, how I select the main camera, and how to get extrinsic of other cameras relative to the main camera

such as:
[image]

I would like to know how to get the external reference of camera k4a2982 with respect to camera k4a3495

Thanks!!!

ankarako, 2022-09-01:
Hi @terry-r123 ,

Unfortunately this information is not captured by the system. Every camera is registered individually.

terry-r123, 2022-09-01:
> Hi @terry-r123 ,
> 
> Unfortunately this information is not captured by the system. Every camera is registered individually.

Thanks,  @ankarako .
Can I make cameras horizontally to calibrate, because we want to shoot video horizontally to widen the field of view

closed 2022-09-14 by ankarako